PX4/PX4-Autopilot #13282: In RTL mode camera triggering does not stoped

moreba1, 2019-10-27:
**Describe the bug**
When flight mode changes from mission to RTL mode camera triggering does not stoped.

**Expected behavior**
When flight mode changes from mission to other modes a stop triggering command must send to vehicle , also when vehicle comes in mission line again camera triggering must be started.

Antiheavy, 2019-10-28:
Yes. This is quite annoying. We’ve added a “camera off” command to the fixed wing landing pattern in QGC, but that is a bit of a hack that only works with RTL_TYPE=1 and isn’t totally reliable.

Antiheavy, 2019-10-28:
Sometimes I’ve wondered if the camera trigger modes (distance, time, etc) should somehow be mavlink parameters associated with the waypoints themselves. That way if you aren’t tracking a waypoint that is a camera trigger one, then the camera will be off.

moreba1, 2019-10-28:
so it must be implemented in all RTL types also in loiter (pause) and LAND mode .

Antiheavy, 2019-10-28:
Yes. Some time ago someone added this type of functionality to Loiter/Hold/Pause mode,  it it too was sort of a hack and does get confused sometimes.

ndepal, 2019-10-28:
I thought leaving mission mode would issue a trigger pause command? Is this not the case anymore or does it not work as intended?

moreba1, 2019-12-20:
This problem was not solved .

closed 2020-12-22 by moreba1