PX4/PX4-Autopilot #13291: Battery failsafe does not cause DESCEND if local position not available

RomanBapst, 2019-10-28:
**Bug description**
Once the battery level reaches critical value the system tries to execute an RTL or LANDING, whichever is configured. The same process is repeated once the battery level reaches emergency level.
However, if no local position is available (altitude is available) then no failsafe will be executed, even if the battery is totally drained.
https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L343

**Alternative**
How about handling this case similar to when the vehicle loses RC signal and no local position is available (altitude is available)? In that case the system does a controlled descend.

**Screenshot from SITL testing on lastest master**
- allowed SITL to drain battery to 0
- enabled low battery failsafe (land)
- stopped gpssim such that only altitude but not xy position is valid

[image]

RomanBapst, 2019-10-28:
@julianoes I'm not quite sure if this is somewhat by design or just because battery failsafe and rc loss failsafe are handled at different locations in the commander.
The RC loss failsafe sets a new nav state while the battery failsafe changes the internal state of the commander.

julianoes, 2019-10-28:
> just because battery failsafe and rc loss failsafe are handled at different locations in the commander.

Yes that's probably why!

dagar, 2019-10-28:
The dream would be to capture all of these details into a single state chart.

PR #13294 (2019-11-05): commander: fix battery failsafe without GPS
This fixes the battery failsafe for the following corner cases:
- Battery failsafe set to Return but we can't do RTL because we don't have a global position or home position. In this case we now switch to   Land. Land might end up in Descend in the failsafe state machine later.
- Battery failsafe set to Land but we can't land because we don't have a   local position. In this case we switch to land anyway and then fall back to descend in the failsafe state machine later.

The "fix" involves ignoring using the main_state_transition and implementing the [... cut]
closed 2019-11-05 by julianoes