PX4/PX4-Autopilot #13292: BAT_A_PER_V und BAT_V_DIV missing

julianoes, 2019-10-28:
**Describe the bug**
When I open the Power Tab in QGC 3.5.4 against PX4 SITL (8f910f84358b0da6ad79231080c63f3542408bc1) I get the warning:

"Parameters are missing from firmware. You may be running a version of firmware QGC does not work correctly with or your firmware has a bug in it. Missing params: -1:BAT_A_PER_V, -1:BAT_V_DIV"

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL
2. Open QGC and go to Settings -> Power tab.

**Expected behavior**
No warning should be displayed.

**Log Files and Screenshots**

[image]

julianoes, 2019-10-28:
```
6a0f5249f825effc6a516596aea7ccc4b8c70cc2 is the first bad commit
commit 6a0f5249f825effc6a516596aea7ccc4b8c70cc2
Author: Daniel Agar <daniel@agar.ca>
Date:   Mon Oct 21 13:40:23 2019 -0400

    sensors: split out analog battery handling to new standalone battery_status module
```

dagar, 2019-10-28:
Short term we can shift the analog battery parameters so that they're still present in SITL (which has no analog batteries).

I think we need to maintain a list of params QGC depends on.

dagar, 2019-10-28:
Longer term we'll need to do something QGC side for vehicles that only have digital power modules or even v5x that's somewhere in between (cell configuration, but no ADC to configure).

dagar, 2019-10-28:
This also makes me wonder about automated testing between QGC and actual PX4 rather than just a mock link.

dagar, 2019-10-28:
@DonLakeFlyer / @dogmaphobic can we start thinking about how to update the QGC power configuration gui for future systems that don't have analog power modules to configure?

dagar, 2019-10-28:
Going into v1.10 - https://github.com/PX4/Firmware/pull/13301

PR #13299 (2019-10-28): sensors: own BAT_V_DIV and BAT_A_PER_V params
 - this is currently necessary for the QGC Power setup gui, but should
be reverted in the future
 - fixes #13292
commit 94de12ef (2019-10-28): sensors: own BAT_V_DIV and BAT_A_PER_V params (#13299)

 - this is currently necessary for the QGC power setup gui, but should be reverted in the future
 - fixes #13292
commit 5bd5fa34 (2019-10-28): sensors: own BAT_V_DIV and BAT_A_PER_V params (#13299)

 - this is currently necessary for the QGC power setup gui, but should be reverted in the future
 - fixes #13292
closed 2019-10-28 by dagar