PX4/PX4-Autopilot #13313: MC Landing Gear Don't Deploy at Correct Altitude at End of Mission

mwiatt, 2019-10-30:
**Describe the bug**
Landing gear are enabled via a mixer to AUX1. Landing gear properly retract upon takeoff. When landing, the gear do not start to deploy until the copter is around 1m above the ground.

**To Reproduce**
Steps to reproduce the behavior:
1. Power on and wait for GPS lock
2. Uploaded mission. Mission End is set to "Return to Launch" [gear-test.txt](https://github.com/PX4/Firmware/files/3786802/gear-test.txt)
3. Took off in mission mode
4. Verified during testing at two locations, 4 flights each.

**Expected behavior**
During the landing phase, landing gear should be commanded to deploy when RTL_DECEND_ALT, LAND_ALT_1, or LAND_ALT_2 is reached though I am not sure which it is (but would like to know). 

**Log Files and Screenshots**
Flight Log showing a very quick flight with one waypoint:
https://logs.px4.io/plot_app?log=c57c04e0-a6bd-4f04-9fe2-eb986cbcc1b4

**Drone (please complete the following information):**
- Tarot X6 (with Tarot retracts)
- Pixhawk Cube 2.1
- Here2 GNSS
- PX4 1.9.2
- QGC 3.5.5

**Additional context**
A file named lgear.aux.mix is placed in /fs/microsd/etc/mixers. This Tarot landing gear controller requires an PWM output of less than 1520 for the gear to retract and greater that 1520 for the gear to extend which is the reason for the (-) scaling. It has the following content:
```
M: 1
S: 0 7 -10000 -10000 0 -10000 10000
```
A file named config.txt is placed in /fs/microsd/etc with the following content:
```
set MIXER_AUX lgear
set PWM_AUX_OUT 1234
set PWM_AUX_DISARMED 1500
set PWM_AUX_MIN 1000
set PWM_AUX_MAX 2000
set PWM_AUX_RATE 50
```
The gear are working properly in mission mode on takeoff as well as manual mode using an RC channel assigned by RC_MAP_GEAR_SW.

mwiatt, 2019-11-03:
The log posted above with generated with a mission that had Mission End set to "Return to Launch". I ran a test flight yesterday without the "Return to Launch" and added a discrete "Land" waypoint at the end of the mission. As before, the landing gear did not deploy at the end of the mission until the copter was a few feet off the ground.

mwiatt, 2019-11-03:
I am pretty sure I found the bug but I am a newbie with PX4 and having some difficulty navigating through the code.

When a waypoint of type "land" is processed, there is a call to _generateLandSetpoints() to update the setpoint:

https://github.com/PX4/Firmware/blob/045f6233d4e77de87a2ef9ffe4e81b26bf9c50eb/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L76

The gear state is updated in _generateLandSetpoints() here:

https://github.com/PX4/Firmware/blob/045f6233d4e77de87a2ef9ffe4e81b26bf9c50eb/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L130

After _generateLandSetpoints() is called, the gear state is updated "if (_highEnoughForLandingGear())" and the "GEAR_DOWN" state is overwritten:

https://github.com/PX4/Firmware/blob/045f6233d4e77de87a2ef9ffe4e81b26bf9c50eb/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L98

The GEAR_DOWN state will only "stick" once _highEnoughForLandingGear() is false which is hardcoded to 2m

https://github.com/PX4/Firmware/blob/045f6233d4e77de87a2ef9ffe4e81b26bf9c50eb/src/lib/FlightTasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L181

MaEtUgR, 2019-11-07:
I'm reviewing your changes and followed back where the above 2 meters altitude check comes from in case you're interested.
The first occurance of an altitude dependent automatic landing gear triggering comes from
https://github.com/PX4/Firmware/commit/d7683e97a30949f65287d5fa70c8abff161ffd99#diff-fc77c3ef569029d45764664c75d4b0c1R1529
and it was changed to the 2 meters you're seeing today in
https://github.com/PX4/Firmware/commit/bc406a122e35abcb23c075df56b0f279b18f95f2#diff-fc77c3ef569029d45764664c75d4b0c1R1569

FlavioTonelli, 2020-09-16:
The linked pull request is merged. This issue shall be fine to close, @mwiatt isn't it?

closed 2020-09-18 by bresch