PX4/PX4-Autopilot #13329: CUAV GPS V5 mag not detected

julianoes, 2019-10-31:
**Describe the bug**
The CUAV GPS V5 magnetometer connected to a Drotek Pixhawk 3 Pro or a Pixracer does not seem to get detected.

**To Reproduce**
1. Connect GPS/mag to a Pixracer.
2. Check `listener sensor_mag` to check if the device shows up.

**Log Files and Screenshots**

```
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: a6a88774637c745ee069c6482a79b7d4b3cc8b12
FW version: 1.10.0 80 (17432704)
FW git-branch: master
OS: NuttX
OS version: Release 7.29.0 (119341311)
OS git-hash: d8da511082646d83a54c6905daca13f0a1a609f0
Build datetime: Oct 31 2019 10:59:45
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000100000000303833363335510d00300041
MCU: STM32F42x, rev. 3
nsh: mount: mount failed: No such device
nsh: mkfatfs: mkfatfs failed: No such device
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [tune_control] Publishing standard tune 16
Board defaults: /etc/init.d/rc.board_defaults
WARN  [safety_button] disabled by CBRK_IO_SAFETY, exiting
ERROR [safety_button] Task start failed (-1)
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
WARN  [rgbled] no RGB led on bus #1
WARN  [rgbled_ncp5623c] no RGB led on bus #1
nsh: rgbled_pwm: command not found
Board sensors: /etc/init.d/rc.board_sensors
reset done, 50 ms
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
INFO  [ist8310] no device on bus 1
BMP280_I2C on I2C bus 1 at 0x76 (bus: 100 KHz, max: 100 KHz)
WARN  [bmp280] id of your baro is not: 0x58
WARN  [bmp280] bus option number is 2
ERROR [bmp280] driver start failed
WARN  [qmc5883] no device on bus 1 (type: 2)
BMM150 on I2C bus 1 at 0x10 (bus: 100 KHz, max: 100 KHz)
WARN  [bmm150] id of magnetometer is not: 0x32
ERROR [bmm150] driver start failed
MS5611_SPI on SPI bus 2 at 3 (20000 KHz)
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
WARN  [hmc5883] no device on bus 1 (type: 3)
WARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 7 (11000 KHz)
WARN  [lis3mdl] no device on bus 3
WARN  [bmi055] No BMI055 gyro found
WARN  [mpu6000] no device on bus #3 (SPI1)
WARN  [mpu6000] no device on bus #3 (SPI1)
WARN  [bmi055] No BMI055 accel found
INFO  [mpu9250] Bus probed: 3
MPU9250 on SPI bus 1 at 4 (1000 KHz)
ERROR [battery_status] no ADC found: /dev/adc0 (2)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
ERROR [mavlink] DM_KEY_MISSION_STATE lock failed
ERROR [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: /fs/microsd/log

NuttShell (NSH)
nsh> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [frsky_telemetry] Scanning timeout: exiting
```

```
nsh> listener sensor_mag

TOPIC: sensor_mag
 sensor_mag_s
        timestamp: 264700518  (0.000555 seconds ago)
        error_count: 0
        device_id: 263178 (Type: 0x04, SPI:1 (0x04)) 
        x: 0.3280
        y: 0.3280
        z: 0.2557
        temperature: 45.1921
        scaling: 0.0015
        x_raw: 186
        y_raw: -186
        z_raw: 150
        is_external: False
```

PX4 version: a6a88774637c745ee069c6482a79b7d4b3cc8b12

julianoes, 2019-10-31:
It's also not working with v1.9.2, and I could not find docs about what device to expect on I2C.

cuhome, 2019-10-31:
Have you tested this compass with CUAV V5+？

cuhome, 2019-10-31:
Does neo gps work for rgb? I can't find these two flight controllers in China, but I tested the pixhawk series flight control with no compatibility issues. For the fmu v4 firmware, I tested the pixhack nano (isn't v5 nano; it has been discontinued; it belongs to fmu v4) It can support ist8310 (neo gps compass); so I think the connection may be wrong.

julianoes, 2019-10-31:
I tried `i2cdetect` and nothing showed up. Also, the GPS doesn't seem to be working either.

> Have you tested this compass with CUAV V5+？

I have not yet.

cuhome, 2019-11-01:
Can you provide me with a connection picture? I will check it.

julianoes, 2019-11-01:
@cuhome can you provide me the pinout of the CUAV V5 GPS/Compass? Is it not compatible with Pixracer?
[image]

cuhome, 2019-11-04:
http://doc.cuav.net/flight-controller/v5-autopilot/en/v5+.html#connection；This is the complete v5 definition, you need to look at the GPS&safety interface; it should not be incompatible with pixracer.

cuhome, 2019-11-04:
[image]

cuhome, 2019-11-04:
@julianoes Your pinouts order is wrong. This picture is the correct guide. Please follow it to fix it.

cuhome, 2019-11-04:
I will add pinouts information for V5+

cuhome, 2019-11-05:
@julianoes Has this problem been solved? If yes, please help me close it.

julianoes, 2019-11-05:
@cuhome so I am supposed to use the colors like you suggest? I don't understand this because I don't have blue and grey but I have orange and yello. So what should I do?

cuhome, 2019-11-06:
Oh, NO, you are using m8n GPS instead of neo GPS, you can modify it according to the following picture;
[image]

julianoes, 2019-11-06:
Aha, why is this not like the reference standard but different? :astonished: Are the future products properly compatible at least?

julianoes, 2019-11-06:
Ok, the GPS works again but the mag does not answer on I2C. Presumably it is destroyed because 5v was applied to SCL or SDA.

julianoes, 2019-11-06:
@Tony3dr @dannyfpv can you check that the wiring on your CUAV GPS is correct?

cuhome, 2019-11-06:
@julianoes Because it is used in the pixhack series, for some reason, the CUAV Pixhack series of flight control does not use the Pixhawk standard pin. What you need to know is that the CUAV M8N GPS will be discontinued. We recommend using neo GPS.

julianoes, 2019-11-06:
Ok, understood, so this is fixed for future products which is good!

closed 2020-02-04 by julianoes