PX4/PX4-Autopilot #13334: SITL tests safe landing intermittent failure

dagar, 2019-10-31:
- http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests_ASan/detail/master/878/pipeline

I'll add more examples and output as these appear.

TSC21, 2019-11-02:
I have no clear idea of what's causing this but looking at the build logs I also see a poll timeout on the mavlink interface. @mrivi any ideas?

dagar, 2019-11-02:
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-13337/6/pipeline

mrivi, 2019-11-04:
this is seems more like a real failure. I need to look into this.

mrivi, 2019-11-04:
I cannot see the CI logs. The link sends me to a white empty jenkins page

julianoes, 2019-11-04:
> I cannot see the CI logs. The link sends me to a white empty jenkins page

Try a couple of times or hit Ctrl+R. Jenkins seems to be that way :sob:.

julianoes, 2019-11-06:
This is apparently still happening.

dagar, 2019-11-06:
Here's a new one. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master/1636/pipeline

[image]

https://logs.px4.io/plot_app?log=dfa49546-b53e-4c85-9ef9-bdd202e6099f

julianoes, 2019-11-07:
@dagar but that's something new. That's not the poll error.

julianoes, 2019-11-07:
@mrivi can you check this one?

julianoes, 2019-11-07:
What about this one?
http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-13405/1/pipeline

dagar, 2019-11-09:
Here's a new one (no poll timeout). http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-13420/2/pipeline

julianoes, 2019-11-11:
@dagar would this matter?

```
`ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
```

dagar, 2019-11-28:
Here's a fresh one. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-13629/1/pipeline

https://logs.px4.io/plot_app?log=d324dadd-6988-41d1-972f-1789704b5891

[image]

[image]

julianoes, 2020-03-02:
I haven't seen this in a while? Is this still an issue @dagar @mrivi ?

PR #13390 (2019-11-05): mixer_module: fix poll error in SITL lockstep
This fixes the case where the mixer_module would subscribe and use its own test_motor publication which was created only to make sure the topic is advertised and subsequent updates will work properly.

This happened in SITL lockstep because the timestamp would be 0 at the very beginning, and hence elapsed time would be 0 as well.
This lead to an actuator publication which would then get lockstep out of sync causing poll errors on the Gazebo side.

This fixes #13377 and maybe fixes #13334 and maybe closes #13371.
closed 2020-06-02 by julianoes