PX4/PX4-Autopilot #13352: Hard-fault produced when GPS is driver is started in-flight

nicovanduijn, 2019-11-01:
**Describe the bug**
This may not be a regular use-case, but while testing to reproduce #13350, I had a bad crash, which turns out was a hard-fault.

**Steps taken**
Steps to reproduce the behavior:
1. Drone with optical flow (pmw3901), GPS actively used. Commanded via MAVLink
2. Stopped the GPS driver
3. Take-off
4. `gps start -b 115200`
5. BOOM

**Elaboration**
I mention that the vehicle was commanded via MAVLink because I did the exact same test again immediately afterwards, with the only difference that I was using an RC and it did not hard-fault. I was scared to attempt to reproduce... Joysticks don't let you map a kill switch!

**Log Files and Screenshots**
[Log](https://review.px4.io/plot_app?log=b3af87ef-bc79-4795-be5e-6c99e36557a9)
[Hard-fault file](https://drive.google.com/file/d/1IxdO0bjZCKV5n7AkUgmThaM9rMiLva93/view?usp=sharing)
[ELF file](https://drive.google.com/file/d/1P0VStSh6af39KoXZ0IefE6NJOosAE7jF/view?usp=sharing)

Antiheavy, 2019-11-01:
Scary.  We've experimented with stopping and starting the GPS driver in flight and have not had issues, but this was back on v1.8.0 based PX4.  Unfortunately our log file got deleted from PX4 review, but the discussion is here:
https://discuss.px4.io/t/neat-experiements-with-fixed-wing-loss-of-gps/7193

julianoes, 2019-11-04:
I just ran `info line *0x...` for anything I found in the hard fault log, and only see this:

```
(gdb) info line *0x08004e1d
Line 2920 of "chip/stm32_serial.c" starts at address 0x8004e1c <up_putc+40> and ends at 0x8004e28 <stm32_pwr_enablebkp>.
(gdb) info line *0x36353936
No line number information available for address 0x36353936
(gdb) info line *0x08004669
Line 251 of "/home/nico/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/include/arch/armv7-m/irq.h" starts at address 0x8004668 <up_restoreusartint+16>
   and ends at 0x800466e <up_restoreusartint+22>.
(gdb) info line *0x0813b8d3
No line number information available for address 0x813b8d3
(gdb) info line *0x20003680
No line number information available for address 0x20003680 <g_intstackalloc+384>
(gdb) info line *0x0813b8f9
No line number information available for address 0x813b8f9
(gdb) info line *0x20003664
No line number information available for address 0x20003664 <g_intstackalloc+356>
(gdb) info line *0x2000361c
No line number information available for address 0x2000361c <g_intstackalloc+284>
(gdb) info line *0x0800f969
Line 202 of "stdio/lib_libvsprintf.c" starts at address 0x800f960 <lib_vsprintf+48> and ends at 0x800f970 <lib_vsprintf+64>.
(gdb) info line *0x20003614
No line number information available for address 0x20003614 <g_intstackalloc+276>
(gdb) info line *0x0000003d
No line number information available for address 0x3d
(gdb) info line *0x08006b85
Line 61 of "syslog/syslog_emergstream.c" starts at address 0x8006b84 <emergstream_putc> and ends at 0x8006b8a <emergstream_putc+6>.
(gdb) info line *0x08006b91
Line 75 of "syslog/syslog_emergstream.c" starts at address 0x8006b90 <emergstream_putc+12> and ends at 0x8006b94 <emergstream_putc+16>.
(gdb) info line *0x20003614
No line number information available for address 0x20003614 <g_intstackalloc+276>
(gdb) info line *0x0000000a
No line number information available for address 0xa
(gdb) info line *0x0000000a
No line number information available for address 0xa
(gdb) info line *0x08004e1d
Line 2920 of "chip/stm32_serial.c" starts at address 0x8004e1c <up_putc+40> and ends at 0x8004e28 <stm32_pwr_enablebkp>.
(gdb) info line *0x0000000a
No line number information available for address 0xa
(gdb) info line *0x08004669
Line 251 of "/home/nico/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/include/arch/armv7-m/irq.h" starts at address 0x8004668 <up_restoreusartint+16>
   and ends at 0x800466e <up_restoreusartint+22>.
```

So it could have something to do with UART configuration when the GPS driver starts?

@davids5 your input would be appreciated.

davids5, 2019-11-04:
I would attach the debugger and put a break point on up_hardfault and step out unitl you will be at the faulting line. If that is no conclusive  I would guess it is a stack crash of the GPS onto what is below it.  => Increase the GPS app and thread's stack sizes and retest.

julianoes, 2020-12-02:
I just tried to reproduce this on latest master, and I can't make it happen anymore. I tried around 30 times. Closing.

closed 2020-12-02 by julianoes