PX4/PX4-Autopilot #13366: v1.10.0 Release Notes + Changelog

mrpollo, 2019-11-03:
It's that time again. We need to band together to bring forward the best release the community can give. Help us represent your work better to the world if you have a feature you worked endless hours, and you think it should be part of the high-level release notes, please let us know in a comment below.

Please review the changelog and help us identify anything that should be part of the high-level release notes.

For reference, here are the v1.9 release notes.

https://github.com/PX4/Firmware/releases/tag/v1.9.0

v.1.10.0 Changelog
https://gist.github.com/mrpollo/3af778f7f6cdbad1f0433ba18b4b7204

bkueng, 2019-11-04:
@mrpollo what is the difference to this doc: https://docs.google.com/document/d/1n6agfdgzk0SfvW61l3vbsMTBj7nT-JIaKsf5hQFs69E/edit?pli=1?

mrpollo, 2019-11-04:
Goot catch @bkueng, as of now, this issue is the source of truth, we can port over any missing info from the doc. 

cc @jinger26

mrpollo, 2019-11-04:
From @davids5 

```
The release includes a change from NuttX-7.28 NuttX-7.29
See NuttX release notes Release Notes for the details of changes in NuttX-7.29 details.
------------------------
Included as back ports are:
STM32:F7 i2C fixes
 State and interrupt storm fixes.
Serial Single wire mode has moved to NuttX
Fix for "z" in printf
kinetis:i2c transfer ensure correct result returned.        kinetis_i2c_transfer released the mutex then fetched
      the state, this resulted in returning the correct
      results.
STM32F4 and STM32F7 SD card hang on card eject fixed
stm32f7:ethernet add timeout on MAC reset```

mrpollo, 2019-11-04:
@davids5 I updated the release notes per your comment on Slack, thanks for your help.

Can you please make sure I got the links right?

davids5, 2019-11-04:
@mrpollo - Thank you. minor changes - Done

RomanBapst, 2019-11-05:
@mrpollo I updated the vtol section.

bkueng, 2019-11-05:
Thanks @mrpollo

MaEtUgR, 2019-11-05:
@mrpollo Nice! I added a comment about TPA for migration.
**EDIT:** and mag inconsistency check improvement.

julianoes, 2019-11-07:
@dagar I think we should mention all the VSCode support.

ThomasRigi, 2019-11-14:
@mrpollo can you add #12778 to the failsafe bugfixes?

mrpollo, 2019-11-14:
Thanks @ThomasRigi I added your fix to the release notes.

mrpollo, 2020-03-11:
Hey Mr stable bot, how timely of you. ⏱

closed 2020-03-11 by mrpollo