PX4/PX4-Autopilot #13377: [SITL] Headless Gazebo simulation broken

ThomasRigi, 2019-11-04:
**Describe the bug**
Running Gazebo in headless mode works only once per PC start. After that, the following error occurs:

[Err] [gazebo_mavlink_interface.cpp:1155] poll timeout
Trying to exit with Ctrl-C throws:
^CERROR [px4] Startup script returned with return value: 2

Problem is observed on v1.10.0-beta4 and current master. I'm on Ubuntu 18.04, @dogmaphobic experienced the same issue on macOS and @RicardoM17 encountered it too.

Running Gazebo with GUI still works, also after experiencing the error with headless mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Start any Gazebo SITL simulation in headless mode, e.g. `HEADLESS=1 make px4_sitl gazebo_standard_vtol`. At the latest after the second time running the command the error appears.
2. See error

**Expected behavior**
No error

**Log Files and Screenshots**
Complete console output:

(base) thomas@thomas-N56VZ:~/src/Firmware$ HEADLESS=1 make px4_sitl gazebo_standard_vtol
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/thomas/src/Firmware/bui...as/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/thomas/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: standard_vtol
src_path: /home/thomas/src/Firmware
build_path: /home/thomas/src/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/thomas/Master_Thesis/PdM_Firmware/build/px4_sitl_default/build_gazebo:/home/thomas/src/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/thomas/Master_Thesis/PdM_Firmware/Tools/sitl_gazebo/models:/home/thomas/src/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/home/thomas/Master_Thesis/PdM_Firmware/build/px4_sitl_default/build_gazebo:/home/thomas/src/Firmware/build/px4_sitl_default/build_gazebo
not running gazebo gui
SITL COMMAND: "/home/thomas/src/Firmware/build/px4_sitl_default/bin/px4" "/home/thomas/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/thomas/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/thomas/src/Firmware/ROMFS/px4fmu_common -> /home/thomas/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
  SYS_AUTOSTART: curr: 0 -> new: 1040
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CAL_BARO_PRIME: curr: 0 -> new: 6620172
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 5.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  IMU_GYRO_RATEMAX: curr: 0 -> new: 400
  MIS_TAKEOFF_ALT: curr: 2.5000 -> new: 20.0000
  MIS_YAW_TMT: curr: -1.0000 -> new: 10.0000
* MPC_ACC_HOR_MAX: curr: 3.0000 -> new: 2.0000
  MPC_TKO_SPEED: curr: 1.5000 -> new: 1.0000
  MPC_VEL_MANUAL: curr: 10.0000 -> new: 3.0000
  MPC_XY_CRUISE: curr: 5.0000 -> new: 3.0000
  MPC_XY_VEL_MAX: curr: 12.0000 -> new: 4.0000
  MPC_Z_VEL_MAX_DN: curr: 1.0000 -> new: 1.5000
* NAV_ACC_RAD: curr: 2.0000 -> new: 3.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  RTL_TYPE: curr: 0 -> new: 1
  WV_EN: curr: 0 -> new: 1
  FW_AIRSPD_MAX: curr: 20.0000 -> new: 25.0000
  FW_AIRSPD_MIN: curr: 10.0000 -> new: 14.0000
  FW_AIRSPD_TRIM: curr: 15.0000 -> new: 16.0000
* MC_ROLLRATE_P: curr: 0.2000 -> new: 0.3000
  MIS_LTRMIN_ALT: curr: -1.0000 -> new: 10.0000
* MIS_TAKEOFF_ALT: curr: 20.0000 -> new: 10.0000
  MPC_THR_MIN: curr: 0.1200 -> new: 0.1000
* MPC_XY_VEL_D: curr: 0.0160 -> new: 0.0050
  MPC_XY_VEL_I: curr: 0.0200 -> new: 0.2000
* MPC_XY_VEL_P: curr: 0.2000 -> new: 0.1500
* NAV_ACC_RAD: curr: 3.0000 -> new: 5.0000
  NAV_LOITER_RAD: curr: 50.0000 -> new: 80.0000
  VT_F_TRANS_THR: curr: 1.0000 -> new: 0.7500
  VT_ARSP_TRANS: curr: 10.0000 -> new: 16.0000
  VT_MOT_ID: curr: 0 -> new: 1234
  VT_FW_MOT_OFFID: curr: 0 -> new: 1234
Gazebo multi-robot simulator, version 9.11.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

  VT_TYPE: curr: 0 -> new: 2
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.178.77
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
[Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_0_joint] found for channel[0] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_1_joint] found for channel[1] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_2_joint] found for channel[2] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_3_joint] found for channel[3] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_puller_joint] found for channel[4] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [left_elevon_joint] found for channel[5] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [right_elevon_joint] found for channel[6] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [elevator_joint] found for channel[7] joint control active for this channel.
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
INFO  [simulator] Simulator connected on TCP port 4560.
[Msg] Using MAVLink protocol v2.0
  MAV_TYPE: curr: 2 -> new: 22
INFO  [init] Mixer: etc/mixers-sitl/standard_vtol_sitl.main.mix on /dev/pwm_output0
[Err] [gazebo_mavlink_interface.cpp:1155] poll timeout
[Err] [gazebo_mavlink_interface.cpp:1155] poll timeout
^CERROR [px4] Startup script returned with return value: 2
pxh> 

**Additional context**
Add any other context about the problem here.

julianoes, 2019-11-05:
I can't reproduce this on Fedora 30, however, I can see it on macOS 10.14.6.

`git bisect` points to any of these commits:
```
<fill in>
```

julianoes, 2019-11-05:
@bkueng it looks like something of the refactoring messed with lockstep.

PR #13390 (2019-11-05): mixer_module: fix poll error in SITL lockstep
This fixes the case where the mixer_module would subscribe and use its own test_motor publication which was created only to make sure the topic is advertised and subsequent updates will work properly.

This happened in SITL lockstep because the timestamp would be 0 at the very beginning, and hence elapsed time would be 0 as well.
This lead to an actuator publication which would then get lockstep out of sync causing poll errors on the Gazebo side.

This fixes #13377 and maybe fixes #13334 and maybe closes #13371.
closed 2019-11-05 by dagar