PX4/PX4-Autopilot #13378: FMUK66 - compass inconsistent

igalloway, 2019-11-04:
**Describe the bug**
This issue was reported directly, i'm cc'ing here. 

> "Thanks for your great work, i got everything work of my RDDRONE-FMUK66 but not the Compass.
> 
> Missionplanner shows "bad compass health" and all compass settings are unavailable.
> QGroundControl shows "compass calibration inconsistent". 
> 
> FMUK66, holybro pixhawk4 m8n gps.
>  
> Compass calibration worked in QGroundcontrol but no effect, still "compass calibration error".
> P.s.: No Loiter or nogps flightmodes selectable. 
>  
> i was searching through a lot communities, without success."

**Additional context**
The Compass inconsistent problem was introduced maybe 6 months ago. I believe many users disabled their second onboard compass as a result.
It is not clear to me if this issue is considered resolved now or not.

In any case - how can we best get useful data to understand this "compass calibration inconsistent" message? What should i ask the user to do or to report?
Thanks
-iain

LorenzMeier, 2019-11-04:
@bresch Do you have insights there? We really need to put this issue to rest - even if it is a real HW issue and not a "software fail". It is an UX issue and the end user really doesn't care what component is to blame. We at least need to help to diagnose this better and if possible automatically work around it.

@julianoes @bkueng I consider this very frustrating to users, please help to keep an eye on it.

igalloway, 2019-11-11:
Another new response from Eastern Michigan University.

> Hello Iain, 
> I did all the calibrations and set all the orientations as mentioned in the gitbook. Before flying(taking off), you mentioned that we need to arm the drone first, I did that as mentioned in https://nxp.gitbook.io/hovergames/userguide/flying/before-you-fly.
> There are some issues I am facing when trying to "Takeoff", it shows these messages:
> "Please set the orientation and recalibrate"
> When I tried arming the drone:
> 
 
[image]
[image]

igalloway, 2019-11-14:
https://www.hackster.io/contests/hovergames/discussion/posts/5776#challengeNav

Hi, I am using MAVROS on a companion computer and no matter what I do I am getting FCU: Preflight Fail: Compass Sensors inconsistent when I try to arm.

In mavros logs I see:
key: "3D magnetometer"
value: "Fail"
I have tried changing the firmware on the drone but can't seem to get it to connect to QGC when I do that. Been stuck for a long time and any help will be much appreciated. Thanks!

bkueng, 2019-11-18:
Can you provide a log?
Make sure to enable `Sensor comparison` in `SDLOG_PROFILE`, and log from boot (`SDLOG_MODE` = `from boot until disarm`).

julianoes, 2020-02-19:
@igalloway is this still an issue or can we close it?

closed 2020-05-20 by julianoes