PX4/PX4-Autopilot #13404: Make stuck on checking submodule Nuttx

iwishiwasaneagle, 2019-11-06:
**Describe the bug**
Building PX4 for a Pixhawk with `make px4_fmu-v2_default` hangs on `bash Tools/check_submodules.sh platforms/nuttx/NuttX/nuttx`. However this command runs instantly in a fresh shell. Waiting 30+ minutes doesn't help issue.

**Expected behavior**
Make builds PX4 and uploads it to the pixhawk board

**Log Files and Screenshots**
```
 $ make px4_fmu-v2_default       
-- PX4 version: v1.10.0-beta4-158-ge3964d5150
-- PX4 config file: /opt/drone/PX4/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled

```
```
$ bash Tools/check_submodules.sh platforms/nuttx/NuttX/nuttx; echo $!
0
```

htop output
[image]

**Drone (please complete the following information):**
- APM Copter detected as PX4 FMU V2 by QGC

**Additional context**
Add any other context about the problem here.

iwishiwasaneagle, 2019-11-06:
> Add any other context about the problem here.
- Fresh source code from PX4/Firmware repo
- Similar issue to https://github.com/PX4/Firmware/issues/3181 
- I can run `make px4_sitl_default jmavsim` and all other SITL make commands perfectly fine

mcsauder, 2019-11-06:
Hi @iwishiwasaneagle , try running `git submodule update --init --recursive` first, and allow some time for it to complete the nuttx step. I've seen your issue on my machine a handful of times and also had it take 20min+ on occasion even with a good download speed, I'm not sure what causes it to hang.  Report back if that doesn't solve the issue!

dagar, 2019-11-06:
Can you try `make distclean` followed by `make submodulesclean`?

iwishiwasaneagle, 2019-11-06:
> Can you try `make distclean` followed by `make submodulesclean`?

```
$ make distclean
Cleared directory 'Tools/jMAVSim'
Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) unregistered for path 'Tools/jMAVSim'
Cleared directory 'Tools/sitl_gazebo'
Submodule 'Tools/sitl_gazebo' (https://github.com/PX4/sitl_gazebo.git) unregistered for path 'Tools/sitl_gazebo'
Cleared directory 'boards/atlflight/cmake_hexagon'
Submodule 'boards/atlflight/cmake_hexagon' (https://github.com/PX4/cmake_hexagon.git) unregistered for path 'boards/atlflight/cmake_hexagon'
Cleared directory 'cmake/configs/uavcan_board_ident'
Submodule 'cmake/configs/uavcan_board_ident' (https://github.com/PX4/uavcan_board_ident.git) unregistered for path 'cmake/configs/uavcan_board_ident'
Cleared directory 'mavlink/include/mavlink/v2.0'
Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) unregistered for path 'mavlink/include/mavlink/v2.0'
Cleared directory 'msg/tools/gencpp'
Submodule 'msg/tools/gencpp' (https://github.com/PX4/gencpp.git) unregistered for path 'msg/tools/gencpp'
Cleared directory 'msg/tools/genmsg'
Submodule 'msg/tools/genmsg' (https://github.com/PX4/genmsg.git) unregistered for path 'msg/tools/genmsg'
Cleared directory 'platforms/nuttx/NuttX/apps'
Submodule 'platforms/nuttx/NuttX/apps' (https://github.com/PX4/NuttX-apps.git) unregistered for path 'platforms/nuttx/NuttX/apps'
Cleared directory 'platforms/nuttx/NuttX/nuttx'
error: could not lock config file .git/modules/platforms/nuttx/NuttX/nuttx/config: No such file or directory
warning: Could not unset core.worktree setting in submodule 'platforms/nuttx/NuttX/nuttx'
Submodule 'platforms/nuttx/NuttX/nuttx' (https://github.com/PX4/NuttX.git) unregistered for path 'platforms/nuttx/NuttX/nuttx'
Cleared directory 'src/drivers/gps/devices'
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/GpsDrivers.git) unregistered for path 'src/drivers/gps/devices'
Cleared directory 'src/drivers/uavcan/libuavcan'
Submodule 'src/drivers/uavcan/libuavcan' (https://github.com/PX4/uavcan.git) unregistered for path 'src/drivers/uavcan/libuavcan'
Cleared directory 'src/lib/DriverFramework'
Submodule 'src/lib/DriverFramework' (https://github.com/PX4/DriverFramework.git) unregistered for path 'src/lib/DriverFramework'
Cleared directory 'src/lib/ecl'
Submodule 'src/lib/ecl' (https://github.com/PX4/ecl.git) unregistered for path 'src/lib/ecl'
Cleared directory 'src/lib/matrix'
Submodule 'src/lib/matrix' (https://github.com/PX4/Matrix.git) unregistered for path 'src/lib/matrix'
Cleared directory 'src/modules/micrortps_bridge/micro-CDR'
Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/micro-CDR.git) unregistered for path 'src/modules/micrortps_bridge/micro-CDR'
Removing Tools/px4airframes/__pycache__/
Removing build/
Removing msg/tools/__pycache__/
Removing src/lib/parameters/px4params/__pycache__/
```
and it gets stuck on `git submodule update --init --recursive` (from htop) when i run `make submodulesclean`

dagar, 2019-11-06:
After `make distclean` what if you just do `git submodule update --init --recursive --force platforms/nuttx/NuttX/nuttx/` to try and update NuttX directly?

iwishiwasaneagle, 2019-11-06:
> Hi @iwishiwasaneagle , try running `git submodule update --init --recursive` first, and allow some time for it to complete the nuttx step. I've seen your issue on my machine a handful of times and [...]

```
$ git submodule update --init --recursive
Submodule 'boards/atlflight/cmake_hexagon' (https://github.com/PX4/cmake_hexagon.git) registered for path 'boards/atlflight/cmake_hexagon'
Submodule 'cmake/configs/uavcan_board_ident' (https://github.com/PX4/uavcan_board_ident.git) registered for path 'cmake/configs/uavcan_board_ident'
Submodule 'src/drivers/uavcan/libuavcan' (https://github.com/PX4/uavcan.git) registered for path 'src/drivers/uavcan/libuavcan'
Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/micro-CDR.git) registered for path 'src/modules/micrortps_bridge/micro-CDR'
Cloning into '/opt/drone/PX4/boards/atlflight/cmake_hexagon'...
Cloning into '/opt/drone/PX4/cmake/configs/uavcan_board_ident'...
Cloning into '/opt/drone/PX4/platforms/nuttx/NuttX/apps'...
Cloning into '/opt/drone/PX4/platforms/nuttx/NuttX/nuttx'...
^Berror: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/NuttX.git' into submodule path '/opt/drone/PX4/platforms/nuttx/NuttX/nuttx' failed
Failed to clone 'platforms/nuttx/NuttX/nuttx'. Retry scheduled
Cloning into '/opt/drone/PX4/src/drivers/uavcan/libuavcan'...
Cloning into '/opt/drone/PX4/src/modules/micrortps_bridge/micro-CDR'...
Cloning into '/opt/drone/PX4/platforms/nuttx/NuttX/nuttx'...
 error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/NuttX.git' into submodule path '/opt/drone/PX4/platforms/nuttx/NuttX/nuttx' failed
Failed to clone 'platforms/nuttx/NuttX/nuttx' a second time, aborting
```
I managed to run `git clone git@github.com:PX4/NuttX.git` successfully, albeit very slowly. I think it might actually be a network issue...

iwishiwasaneagle, 2019-11-06:
> After `make distclean` what if you just do `git submodule update --init --recursive --force platforms/nuttx/NuttX/nuttx/` to try and update NuttX directly?

```
$ git submodule update --init --recursive --force platforms/nuttx/NuttX/nuttx/   
Cloning into '/opt/drone/PX4/platforms/nuttx/NuttX/nuttx'...
```
and then stays like that for a while... I managed to run `git clone git@github.com:PX4/NuttX.git` successfully, albeit very slowly

mcsauder, 2019-11-06:
I just ran a fresh clone, then `git submodule update --init --recursive`, and it stalled on Nuttx for about 8 minutes but the completed, and then `make px4_fmu-v2_default` and it completes successfully for me, but the Nuttx module took about 8 minutes to complete.

In the past I have stopped the `git submodule update --init --recursive` process and restarted it a few times in a row until it went more quickly, or have waited it out successfully as well.  As long as your terminal responds to `enter` key presses with newlines, you keep waiting for the submodule update to complete, or stop the process (`ctrl-C`), and restart again.  I've had both work.  Let us know if allowing longer to complete the submodule update step for Nuttx works out.

iwishiwasaneagle, 2019-11-06:
> I just ran a fresh clone, then `git submodule update --init --recursive`, and it stalled on Nuttx for about 8 minutes but completed, and then `make px4_fmu-v2_default` and it completed successfully for me.
> 
> In the past I have stopped the `git submodule update --init --recursive` process and restarted it a few times in a row until it went more quickly, or have waited it out successfully as well. As long as your terminal responds to `enter` key presses with newlines, you keep waiting for the submodule update to complete, or stop the process (`ctrl-C`), and restart again. I've had both work. Let us know if allowing longer to complete the submodule update step for Nuttx works out.

Yup spot on! I'm nut used to processes running for 15 minutes + with zero feedback... Cheers for the help!

closed 2019-11-06 by iwishiwasaneagle

mcsauder, 2019-11-06:
Great! Glad it worked!