PX4/PX4-Autopilot #13408: Aircraft not automatically disarming while in Land flight mode

Zaposa, 2019-11-07:
**Describe the bug**
I'm having an issue where my hexacopter will not disarm once the aircraft has landed. The aircraft was flown in Position flight mode (or Stabilize flight mode) and then changed to Land flight mode. The aircraft is able to successfully descend and touch the ground successfully. However, instead of disarming, the motors speed up, which results in the aircraft wanting to tip over and crash.

I have looked on the internet and I have found that it is important to get the hover throttle percentage correct as well as  the manual minimum throttle percentage. I have adjusted these values but still no success. I also changed the parameter (COM_DISARM_LAND) to 0.1 seconds, but still no success.

I know that there isn’t an issue with the actual aircraft as I have managed to successfully perform a mission using APM firmware, where it was able to land and disarm safely.

Is this possibly an issue with the firmware or does someone perhaps know a solution to this problem?

P.S. I have disabled the arming switch (changed parameter CBRK_IO_SAFETY from 0 to 22027)
 
**To Reproduce**
Steps to reproduce the behavior:
1. Disable arming switch
2. Change flight mode to Position or Stabilize
3. Arm & takeoff to an altitude between (5 - 10 metres)
4. Switch flight mode to Land

**Expected behavior**
The aircraft should land and disarm successfully and not want to tip over and crash.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=bc0410c1-4285-4781-bf1b-b6ddee3b120d

**Drone**
DJI F550 Flame Wheel (hexacopter)

julianoes, 2019-11-07:
Thanks for the issue and the log. @bresch do you see what the issue is? Is this expected with 1.9.2?

RodBelaFarin, 2019-11-26:
I have exactly the same problem with my Multicopter. Any updates on this?

bresch, 2020-03-04:
We're planning to review the land detection logic in the near future. Please post logs here if you still have an issue with the land detector.

Antiheavy, 2020-03-04:
> We're planning to review the land detection logic in the near future.

@bresch are you planning to review MC land detector, FW land detector, or both?  (I have comments related to FW land detector if that is with scope of your effort).

bresch, 2020-03-10:
@Antiheavy I'll mainly focus on the MC land detector, but please, share your ideas for FW; I'll keep them in mind if I need to rework the interface.

haihai008, 2020-04-23:
I've had the same problem using APM firmware. Finally, it was found that flight control had installed ultrasonic height detection equipment, and it was normal to restore the default settings.