PX4/PX4-Autopilot #13413: FW: vehicle stays forever in takeoff mode if airspeed fails

sfuhrer, 2019-11-07:
If the airspeed is invalid (eg NAN) during FW mission takeoff (runway takeoff), the take off state is never exited. 

[image]

https://review.px4.io/plot_app?log=ce5a3aee-13a2-4ee6-bd07-8e1f9b7a497a

This will btw also not be fixed with https://github.com/PX4/Firmware/pull/12887 and the default values (fallback to ground-windspeed would need to be disabled), and is part of the reason that the SITL CI fails for this PR till now).

FYI @dagar @RomanBapst

Antiheavy, 2019-11-07:
> If the airspeed is invalid (eg NAN) during FW mission takeoff (runway takeoff), the take off state is never exited.

terrifying.

> This will btw also not be fixed with #12887 and the default values (fallback to ground-windspeed would need to be disabled), and is part of the reason that the SITL CI fails for this PR till now).

@sfuhrer can you clarify, does #12887 fix this issue or not fix this issue?

Antiheavy, 2019-11-07:
fyi @Kjkinney

sfuhrer, 2019-11-08:
> terrifying.
>
>
> @sfuhrer can you clarify, does #12887 fix this issue or not fix this issue?

#12887 fixes it, but not with the default parameters (you need to enable the airspeed checks and fallback to groundspeed-windspeed as airspeed estimation). 

Independently of what #12887 does though this issue needs to be addressed inside the Runway takeoff logic. The problem there is that it is currently always requiring an airspeed measurement. If it doesn't, it results in the flyaway reported above (also with FW_ARSP_MODE set to 1, so disabled airspeed). We need to discuss what the desired behavior is: Abort the takeoff or switch to a time-based logic (similar to what the VTOL transition logic does in case of no airspeed data).

sfuhrer, 2021-01-18:
Airspeed failure detection not enabled by default: fixed by https://github.com/PX4/PX4-Autopilot/pull/16564

julianoes, 2021-02-09:
@sfuhrer is still an issue?

closed 2021-02-09 by sfuhrer