PX4/PX4-Autopilot #13415: Violent jerk when switching from manual to position mode

dakejahl, 2019-11-07:
**Firmware**: 1.9.2 (with lidar lite fixes and kill on disarm back ported)
**Airframe**: IF750

**Describe the bug**
I'm seeing huge pitch/roll setpoints when switching from manual to position mode . It seems like it's the velocity controller, but I don't really understand how the velocity controller ends up demanding such large effort of the attitude controller. I've detuned the P gain on the velocity controller and am still seeing a violent jerk when switching from manual to position. Here's a log with 3 mode switch, the first one being the most violent.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch from `manual` to `position` mode. The more velocity you have before the switch, the more violent the jerk.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=4fa7056e-d1bc-437f-829d-0b45e3087276
[image]
[image]

The velocity setpoints are not changing dramatically, so the attitude effort should not be there.
[image]

@bresch @mhkabir @TSC21 

Related to #12389

mhkabir, 2019-11-07:
Any chance you can try v1.10?

bresch, 2019-11-18:
@mhkabir  It's also on master. I found out that the problem is in the initialization of the velocity controller, check on the following graph, the difference between a transition from stabilized -> position and altitude -> position:
[image]
The Z thrust (thrust[1]) initializes at a high value.
After discussing with @MaEtUgR , he remembered that the numerical derivative used in the D-term of the velocity controller could be badly initialized.

I then ran a new simulation with the D gain set to 0 and there is no glitch anymore:
[image]

@MaEtUgR Is already working on the refactoring of the position controller and will make a PR asap. I think we should also backport this fix in v1.10

closed 2019-11-19 by bresch