PX4/PX4-Autopilot #13416: Please I need help here...I have tried running "make px4_sitl gazebo_solo". I keep having this "[px4_sitl] Error 1"

AstonGemmy, 2019-11-08:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "LiftDragPlugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[1/6] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/cornell/Documents/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/cornell/Documents/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

julianoes, 2019-11-08:
Please share version of PX4, version of your Ubuntu, etc. and paste the full build output for:
```
make clean
make px4_sitl gazebo_solo
```

TSC21, 2019-11-08:
You don't have gstreamer1.0 installed.

AstonGemmy, 2019-11-08:
VERSION

OS version is Ubuntu 18.04.3 LTS

PX4 version is v1.9.2

OUTPUT of make clean and make px4_sitl gazebo_solo

cornell@cornell:~/Documents/Firmware$ make clean
cornell@cornell:~/Documents/Firmware$ make px4_sitl gazebo_solo
-- PX4 version: v1.10.0-beta4-180-gb943bd72ab
-- PX4 config file: /home/cornell/Documents/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-88-gd76b704
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cornell/Documents/Firmware/build/px4_sitl_default
[192/714] git submodule src/drivers/gps/devices
[197/714] git submodule src/lib/ecl
[203/714] git submodule mavlink/include/mavlink/v2.0
[204/714] git submodule Tools/sitl_gazebo
[210/714] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
[... 125 lines cut ...]
    linked by target "gazebo_wind_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[255/714] Building CXX object src/lib/.../ecl_EKF.dir/range_finder_checks.cpp.o
FAILED: cd /home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/cornell/Documents/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/cornell/Documents/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[255/714] Building CXX object src/lib/...MakeFiles/ecl_EKF.dir/ekf_helper.cpp.o
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
cornell@cornell:~/Documents/Firmware$

AstonGemmy, 2019-11-08:
> You don't have gstreamer1.0 installed.

How do I get this done please?
Thanks

TSC21, 2019-11-08:
> > You don't have gstreamer1.0 installed.
> 
> How do I get this done please?
> Thanks

Check the README of the sitl_gazebo repo.

AstonGemmy, 2019-11-08:
> Please share version of PX4, version of your Ubuntu, etc. and paste the full build output for:
> 
> ```
> make clean
> make px4_sitl gazebo_solo
> ```

VERSION

OS version is Ubuntu 18.04.3 LTS

PX4 version is v1.9.2

OUTPUT of make clean and make px4_sitl gazebo_solo

cornell@cornell:~/Documents/Firmware$ make clean
cornell@cornell:~/Documents/Firmware$ make px4_sitl gazebo_solo
-- PX4 version: v1.10.0-beta4-180-gb943bd72ab
-- PX4 config file: /home/cornell/Documents/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-88-gd76b704
-- Configuring done
-- Generating done
-- Build files have been written to: /home/cornell/Documents/Firmware/build/px4_sitl_default
[192/714] git submodule src/drivers/gps/devices
[197/714] git submodule src/lib/ecl
[203/714] git submodule mavlink/include/mavlink/v2.0
[204/714] git submodule Tools/sitl_gazebo
[210/714] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
[... 132 lines cut ...]
    linked by target "gazebo_wind_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/cornell/Documents/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[255/714] Building CXX object src/lib/.../ecl_EKF.dir/range_finder_checks.cpp.o
FAILED: cd /home/cornell/Documents/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/cornell/Documents/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/cornell/Documents/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
[255/714] Building CXX object src/lib/...MakeFiles/ecl_EKF.dir/ekf_helper.cpp.o
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
cornell@cornell:~/Documents/Firmware$

AstonGemmy, 2019-11-09:
> > > You don't have gstreamer1.0 installed.
> > 
> > 
> > How do I get this done please?
> > Thanks
> 
> Check the README of the sitl_gazebo repo.

I'm getting this when I try to install Gstreamer

~$ sudo apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris) -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gstreamer1.0-clutter-3.0 is already the newest version (3.0.26-1).
gstreamer1.0-alsa is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-gl is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-gtk3 is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-gtk3 set to manually installed.
gstreamer1.0-packagekit is already the newest version (1.1.9-1ubuntu2.18.04.5).
gstreamer1.0-plugins-base is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-plugins-base-apps is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-plugins-good is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-pulseaudio is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-tools is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-x is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-libav is already the newest version (1.14.5-0ubuntu1~18.04.1).
gstreamer1.0-plugins-bad is already the newest version (1.14.5-0ubuntu1~18.04.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gstreamer1.0-python3-dbg-plugin-loader : Conflicts: gstreamer1.0-python3-plugin-loader but 1.14.5-0ubuntu1~18.04.1 is to be installed
 gstreamer1.0-python3-plugin-loader : Conflicts: gstreamer1.0-python3-dbg-plugin-loader but 1.14.5-0ubuntu1~18.04.1 is to be installed
E: Unable to correct problems, you have held broken packages.

AstonGemmy, 2019-11-09:
It was the GSTREAMER issue actually.
I found my way around it here.[https://discuss.px4.io/t/ubuntu-18-04-gstreamer-cant-be-installed-so-gazebo-video-steaming-doesnt-work/10161/7](url)

I ran the following commands to get the packages.

`sudo apt install libgstreamer1.0-dev`
`sudo apt install gstreamer1.0-plugins-good`
`sudo apt install gstreamer1.0-plugins-bad`
`sudo apt install gstreamer1.0-plugins-ugly`

Thanks @TSC21

closed 2019-11-09 by AstonGemmy

julianoes, 2019-11-11:
Great, thanks for reporting back!

YANG-DEMIN, 2020-03-03:
I am so sorry about that i cant find the  address that you paste. This one:https://discuss.px4.io/t/ubuntu-18-04-gstreamer-cant-be-installed-so-gazebo-video-steaming-doesnt-work/10161/7
And i try to input the comment line that you listed, it does not work.
I want get the solving way, thank you very much!