PX4/PX4-Autopilot #13431: Mission mode error when mission has finished and execute again

SwiftGust, 2019-11-11:
**Describe the bug**
Mission mode execution with remaining mission on the vehicle gives "No valid mission available" error.

**To Reproduce**
Steps to reproduce the behavior:
1. Design some mission on the QGC.
2. Upload the mission.
3. Execute the mission.
4. Land
5. QGC will ask whether delete the mission or leave the mission.
6. Choose leave the mission.
7. Execute the mission.
8. The vehicle will be in loiter with message "No valid mission available. Loitering"

**Expected behavior**
Execute mission properly.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=6a2194e7-ef8e-4a63-8c34-0d43a6dc6b11

julianoes, 2019-11-13:
I just tried to reproduce this in SITL and for it started again with the first mission item after it had landed and I had clicked "Leave the mission".

Does your vehicle disarm when landed or is it still armed? It might only reset the mission item when it disarms.
And note that you can also reset the mission by clicking mission item 1 in QGC.

SwiftGust, 2019-11-13:
Hi, 
This happens time to time, and it disarms after landing.
What I want is vehicle to repeat the same mission executed just before without to have download mission from vehicle & upload mission to vehicle again.

julianoes, 2019-11-13:
What do you mean "time to time"? So it happens sometimes but sometimes it works?

julianoes, 2019-11-13:
@dagar I don't think QGC resets anything. It's a PX4 feature.

All that QGC does is "hideDialog()":

https://github.com/mavlink/qgroundcontrol/blob/d6c3a1eb69e221ca5d92ac5cbde6664566d042f4/src/FlightDisplay/FlightDisplayView.qml#L213-L218

SwiftGust, 2019-11-14:
Yes, This happens occasionally on the vehicle with pixhawk4 onboard when trying to repeate exact same simple mission again.
and sometime there are error & warning message found in the log when successfully performed mission
when starting
`[commander] Takeoff detected`
`[commander] Arming denied! Not landed`

when finished
`[navigator] mission check failed`
`[navigator] No valid mission available, refusing takeoff.`

julianoes, 2019-11-14:
> Right, I was suggesting that QGC should probably be (re-)setting the first waypoint as part of "Start Mission".

I think the existing behavior where px4 does that is fine, it just needs to always work.

julianoes, 2020-02-14:
I would argue that as long as you always disarm, it should work as expected and start at 0 again. If you don't disarm on ground, then this might happen.

ryanjAA, 2023-01-13:
@julianoes could this be related? Started a mission then aborted the takeoff then it said mission finished, clicked keep and then when I tried the mission again this log occurred (went into loiter when I put it in mission and it took off). 

https://discord.com/channels/1022170275984457759/1062041687062487171/1063212114002579496