PX4/PX4-Autopilot #13434: PX4 SITL memory leak

lovettchris, 2019-11-11:
Leave PX4 SITL running over night and the next morning it is using 1.2 Gigabytes of RAM.

**To Reproduce**
Steps to reproduce the behavior:
1. make px4_sitl_default none_iris
2. do nothing else for 12 hours
3. bring up task manager and take a look at the process:

[image]

If you try and connect a simulator you get this error:
ERROR [simulator] Creating TCP socket failed: No buffer space available

[image]

dagar, 2019-11-11:
Thanks Chris. Based on what's running here and how noticeable this would be on an embedded platform (I routinely review memory usage throughout real flights) I would first suspect a leak in either the px4_daemon or simulator module.

px4_daemon: https://github.com/PX4/Firmware/tree/master/platforms/posix/src/px4/common/px4_daemon
Simulator module: https://github.com/PX4/Firmware/tree/master/src/modules/simulator

dagar, 2019-11-11:
It's running overnight before any simulator is connected?

Leaking somewhere stuck in the initial connect perhaps? https://github.com/PX4/Firmware/blob/1c4e854f93d5c90c522d56c905772633be7a9551/src/modules/simulator/simulator_mavlink.cpp#L660-L734

I'll look through it in detail later.

lovettchris, 2019-11-11:
Yeah, I don't normally leave it running like that, just thought you'd be interested in the leak though.  I've been staring at the code you listed, I'm in the TCP connect loop, but I don't see a leak there.  Do you ?

dagar, 2019-11-11:
> I'm in the TCP connect loop, but I don't see a leak there. Do you ?

No, but I was on mobile and hoping someone else could take a closer look. :smirk:

julianoes, 2019-11-13:
Thanks @lovettchris, that looks like fun to debug :smile:!

julianoes, 2019-11-13:
Just a thought: maybe it's something in the lockstep timeout list.

julianoes, 2019-11-18:
Not sure if this is related but it also looks like a leak:

```
HIL_SENSOR received: 3115104349
HIL_ACTUATORS sent: 3115104349, 1574061037822811 (3115104349)
HIL_SENSOR received: 3115107350
HIL_ACTUATORS sent: 3115107350, 1574061037825812 (3115107350)
HIL_SENSOR received: 3115110347
HIL_ACTUATORS sent: 3115110347, 1574061037828809 (3115110347)
HIL_SENSOR received: 3115113346
HIL_ACTUATORS sent: 3115113346, 1574061037831808 (3115113346)
HIL_SENSOR received: 3115116367
HIL_ACTUATORS sent: 3115116367, 1574061037834829 (3115116367)
HIL_SENSOR received: 3115119366
HIL_ACTUATORS sent: 3115119366, 1574061037837828 (3115119366)
HIL_SENSOR received: 3115122347
HIL_ACTUATORS sent: 3115122347, 1574061037840809 (3115122347)
HIL_SENSOR received: 3115125355
HIL_ACTUATORS sent: 3115125355, 1574061037843817 (3115125355)
HIL_SENSOR received: 3115131355
HIL_ACTUATORS sent: 3115131355, 1574061037849817 (3115131355)
HIL_SENSOR received: 3115134358
WARN  [cdev] sim_send: px4_poll() sem error: No error
HIL_ACTUATORS sent: 3115134358, 1574061037852820 (3115134358)
HIL_SENSOR received: 3117417801
ERROR WARN  WARN  [sensors] [commander] [mc_pos_control] Accel #0 fail:  TIMEOUT!Failsafe enabled: no global positionPosition-Control Setpoint-Update failed

ERROR [sensors] WARN  Sensor Accel #0 failed. Reconfiguring sensor priorities.[mc_pos_control]
Failsafe: stop and waitWARN
[sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
HIL_ACTUATORS sent: 3117417801, 1574061040136263 (3117417801)
HIL_SENSOR received: 3119359177
WARN  [px4] px4_sem_wait failure
WARN  [px4] px4_sem_post failure
ERROR [sensors] Gyro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
ERROR [sensors] Mag #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Mag #0 priority: 1
ERROR [sensors] Baro #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Baro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Baro #0 priority: 1
WARN  [cdev] sim_send: px4_poll() sem error: No error
HIL_ACTUATORS sent: 3119359177, 1574061042077639 (3119359177)
HIL_SENSOR received: 3140006329
WARN  [commander] Data link lost
WARN  [ecl/EKF] 3114975314: EKF baro hgt timeout - reset to GPS
WARN  [ecl/EKF] 3114975314: EKF gps hgt timeout - reset to GPS
WARN  [cdev] sim_send: px4_poll() sem error: No error
HIL_ACTUATORS sent: 3140006329, 1574061062724791 (3140006329)
HIL_SENSOR received: 3152335737
WARN  [ecl/EKF] 3114987319: EKF gps hgt timeout - reset to GPS
WARN  [commander] Data link lost
WARN  [cdev] sim_send: px4_poll() sem error: No error
HIL_ACTUATORS sent: 3152335737, 1574061075054199 (3152335737)
HIL_SENSOR received: 3164521030
WARN  [ecl/EKF] 3114993319: EKF gps hgt timeout - reset to GPS
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
WARN  [cdev] sim_send: px4_poll() sem error: No error
ERROR [simulator] timeout waiting for _actuator_outputs_sub
```

julianoes, 2019-11-18:
Also, just running jMAVSim SITL on Windows is leaking about 0.5 MB/s.

lovettchris, 2019-11-18:
Are you saying this leak only happens in my branch or does it also happen in master?  I reported a leak also: https://github.com/PX4/Firmware/issues/13434

julianoes, 2019-11-18:
And I can't reproduce this on Linux.

julianoes, 2019-11-20:
Aha, it does not leak without lockstep!

julianoes, 2019-11-20:
And the leak was already present on v1.9.2.

julianoes, 2019-11-22:
I had the suspicion that something about `thread_local` storage might not be implemented correctly in Cygwin and it looks like that might be true.

We're using `static thread_local` here:
https://github.com/PX4/Firmware/blob/5b235c34c2f1d00e9bfb920656e1cc45f5b20c94/platforms/posix/src/px4/common/lockstep_scheduler/src/lockstep_scheduler.cpp#L65

And if I add a `printf()` in the `~TimedWait()` here:
https://github.com/PX4/Firmware/blob/5b235c34c2f1d00e9bfb920656e1cc45f5b20c94/platforms/posix/src/px4/common/lockstep_scheduler/include/lockstep_scheduler/lockstep_scheduler.h#L22-L23

It never gets called.

According to a question on the cygwin mailinglist in 2014 it seems like non-trivial destructors for `thread_local` storage might indeed not be implemented: http://lists-archives.com/cygwin/85003-destructors-not-called-for-c-11-thread_local-objects.html.

@bkueng do you see an alternative to using this? If not, we should probably try to move to "Bash on Ubuntu on Windows" instead of Cygwin, or just disable lockstep on Windows but that's also suboptimal.

bkueng, 2019-11-22:
> @bkueng do you see an alternative to using this?

It is the best way to implement it (in terms of memory and runtime overhead).
We should not be spawning threads after startup, so I don't see why the usage keeps increasing. Can you check that?

julianoes, 2019-11-22:
> We should not be spawning threads after startup, so I don't see why the usage keeps increasing. Can you check that?

Maybe I'm on a wrong lead then!

julianoes, 2019-11-24:
> Edit: my theory is that it creates a TimedWait object on each wait/iteration.

Ok, I read through the code again. I was wrong.