PX4/PX4-Autopilot #13439: EKF2 External Vision no longer working accurately

nbelanger99, 2019-11-11:
Running 1.9.0 on a Pixhawk 4
Mavros & extras 0.32.2 on Pi 3 B+ running Ubuntu 16.04.6 LTS

We are trying to use an Optitrack for external vision, and we're having some issues. The fused estimate actually seemed accurate and reliable at first when doing some preliminary testing, but it arbitrarily gave out on us when trying to use the position controller and hasn't seemed to work well since. **There were no changes to parameters, hardware, or software.** The local position estimate lags the vision and loosely overshoots and oscillates all over the place even though the vision estimate seems rock solid. It's difficult to explain the behavior exactly, so I've linked a log file and a rosbag of pitching, rolling, yawing, translating in (roughly) a square, and just waving the drone around a bit. 

[2019-11-11_debug.zip](https://github.com/PX4/Firmware/files/3833423/2019-11-11_debug.zip)

dagar, 2019-11-11:
Do you have a log from when it was working well?
Have you tried newer PX4 (eg master or the v1.10 release candidate)?

nbelanger99, 2019-11-11:
Yes, we did try v1.10 and it didn't seem to make a difference. 

I don't have a PX4 log file from when it was working well, but I do have a ROSbag. This bag doesn't show much translation; mostly pitch and roll, but it was fusing and you can see the sort of snappy, accurate estimation that we're used to seeing. All of the aforementioned hardware and software are the same **except**  that we were using a Vive instead of the Optitrack in this test (but we've seen this kind of performance with the Optitrack, and tests have shown its data to be comparable).

mhkabir, 2019-11-11:
Sounds like a timestamping issue to me. Check your offboard timestamps (and timesync), and make sure EKF2_EV_DELAY is set correctly.

closed 2020-01-27 by kamilritz