PX4/PX4-Autopilot #13452: Took off (and crash) even without valid local position

PLMR-FR, 2019-11-12:
**Describe the bug**
In offboard mode, drone was able to arm and take off without valid local position, after taking off, EKF2 became crazy and then result in crash.

**To Reproduce**
Arm and takeoff in offboard mode with "condition_local_position = false"

**Expected behavior**
Drone takes off and flies away

**Log Files and Screenshots**
log: https://review.px4.io/plot_app?log=5d49c71f-0608-4677-b551-70dfcfe7701d

Here is a screenshot with arming state (yellow) and condition_local_position (pink)
[image]

Drone setup:
TBS Discovery
PixHawk FMUV2
Firmware: Master
PX4Flow
EKF2
No GPS

dagar, 2019-11-12:
The offboard messages aren't logged by default. What type of offboard message were you sending?

PLMR-FR, 2019-11-12:
For taking off I send firstly a stream of speed setpoints (with Vx and Vy set to 0) . And then after reaching a certain altitude I switch to a stream of position setpoints for X, Y and Z. 
I did maybe 500 take off with this procedure previously without problems.