PX4/PX4-Autopilot #13453: bad $-escape (literal $ must be written as $$)

473867143, 2019-11-13:
I have some trouble with make **px4_fmu-v5_default**.
**Error:ninja: error: build.ninja:13001: bad $-escape (literal $ must be written as $$)**
os:ubuntu16.04
gcc:5.4.0 and 7.21.
cmake:3.5.1
ninja:1.5.1

```
amov@amov:~/Firmware$ make px4_fmu-v5_default 
-- PX4 config file: /home/amov/Firmware/boards/px4/fmu-v5/default.cmake
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- PX4 version: v1.9.2
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1
-- Building and including px4_io-v2_default
-- Using C++11
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amov/Firmware/build/px4_fmu-v5_default
ninja: error: build.ninja:13001: bad $-escape (literal $ must be written as $$)

Makefile:193: recipe for target 'px4_fmu-v5_default' failed
make: *** [px4_fmu-v5_default] Error 1
```

Can anyone help me ?

julianoes, 2019-11-13:
Have you changed anything? If yes, what?

Has this worked previously? If so, could you do a `git bisect` to check where it broke?

473867143, 2019-11-13:
@julianoes  thanks your re
But, i can build v1.8.2(it's ok). can't only build above v1.9.0.
i don't know the difference between v1.8.2 and v1.9.0.

```
amov@amov:~/Desktop/px4-src/src-v1.8.2/Firmware$ make px4fmu-v5_default 
-- PX4 VERSION: v1.8.2
-- CONFIG: nuttx_px4fmu-v5_default
-- Build Type: MinSizeRel
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-553-g1a11068
-- Building and including px4io-v2
-- Using C++03
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- NuttX: px4fmu-v5 nsh cortex-m7
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amov/Desktop/px4-src/src-v1.8.2/Firmware/build/nuttx_px4fmu-v5_default
ninja: Entering directory `/home/amov/Desktop/px4-src/src-v1.8.2/Firmware/build/nuttx_px4fmu-v5_default'
[9/957] Building px4io-v2
-- PX4 VERSION: v1.8.2
-- CONFIG: nuttx_px4io-v2_default
-- Build Type: MinSizeRel
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-553-g1a11068
-- NuttX: px4io-v2 nsh cortex-m3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/amov/Desktop/px4-src/src-v1.8.2/Firmware/build/nuttx_px4io-v2_default
ninja: Entering directory `/home/amov/Desktop/px4-src/src-v1.8.2/Firmware/build/nuttx_px4io-v2_default'
[193/193] Creating /home/amov/Desktop/..._px4io-v2_default/px4io-v2_default.px4
[957/957] Creating /home/amov/Desktop/...x4fmu-v5_default/px4fmu-v5_default.px4
```

julianoes, 2019-11-20:
@473867143 it would be great if you could do `git bisect` to figure out where it breaks.

473867143, 2019-11-22:
3ks @julianoes

closed 2019-11-22 by 473867143