PX4/PX4-Autopilot #13455: Require valid mission before arm broken in SITL

cmic0, 2019-11-13:
**Describe the bug**
While testing this PR https://github.com/PX4/Firmware/pull/13454 in SITL, I realized I was able to take off without a mission even if `COM_ARM_MIS_REQ` was enabled. 

I checked  `vehicle_status_flags.condition_auto_mission_available` and seems that is always true.
With further analysis i ended up here
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L1652

where seems  that `checkMissionFeasible` reports true.

**To Reproduce**
Steps to reproduce the behavior:
1. Laungh SITL 
2. Clean mission from vehicle
3. Set `COM_ARM_MIS_REQ`to 1
4. Try to arm.

**Expected behavior**
Arming should be denied as no mission is available on the vehicle. 
In reality you can still arm

The issue is only reproducible in SITL. On a real HW works fine

FYI @julianoes

julianoes, 2020-02-14:
@MaEtUgR more pre-arm check things...

MaEtUgR, 2020-10-12:
I cannot reproduce this. Trying SITL jmavsim:
- `vehicle_status_flags.condition_auto_mission_available` correctly represents the mission status when uploading and clearing it.
- `COM_ARM_MIS_REQ` gets taken over correctly into the `_arm_reqwuirements` in commander without reboot
- when running `commander check` on the shell the preflight checks pass because the arm requirements get overridden with no requirements since the command does not have access to the state of commander: https://github.com/PX4/Firmware/blob/2b5b6b943083b92d5870fb2d6a902d4f11a59997/src/modules/commander/Commander.cpp#L309 We should do the check within commander and just pass back the result.

Generally, this issue seems fixed with the caveat that `commander check` does not show the result properly which is a different issue.

closed 2020-10-12 by MaEtUgR