PX4/PX4-Autopilot #13458: Yaw not following ROI in mission

tuloski, 2019-11-13:
Autopilot: Pixracer

Firmware: 1.9.2

I tried many combinations of parameters, and different waypoint (with camera, without camera, ...) but it seems that in no way the multicopter uses the yaw to track a ROI.
I have a gimbal with only pitch, and the pitch seems correct, but no way about the yaw.
Is there a secret combo of parameters, or simply the feature got broken on 1.9.2?

dagar, 2019-11-13:
Could you verify if this is the case in current master or the v1.10 release candidate (https://github.com/PX4/Firmware/releases/tag/v1.10.0-rc1)?

tuloski, 2019-11-14:
I'll test it today.

tuloski, 2019-11-14:
Also the overall mission is working really badly:

- Sometimes it doesn't enter in mission flight mode from RC switch. I have to switch 2-3 times
-Sometimes it doesn't see the waypoint as reached so the quadrotor stops.
-Sometimes it does yaw pirouettes (360 on yaw)

I'll try to revert back to 1.9.1 or 1.8.2.

tuloski, 2019-11-19:
Tested on simulator:

- On 1.9.2 and 1.9.1 it **doesn't work** properly
- On 1.8.2 it's **working** properly

Something got messed up in the middle.

julianoes, 2020-03-02:
FYI @bresch

julianoes, 2020-03-02:
@tuloski can you check 1.10?

tuloski, 2020-03-21:
I tested the latest release and it's working there.

closed 2020-03-21 by TSC21