PX4/PX4-Autopilot #13462: Landed state not updating in mavlink_messages.cpp

julianoes, 2019-11-13:
While working on [testing using MAVSDK](https://github.com/PX4/Firmware/pull/13075) I ran into a problem where the landed state sent via MAVLink telemetry is not up-to-date and made the test fail. Basically on-ground was reported for too long and it didn't switch to takeoff or in-air.

Some debugging showed me that the land_detector was correct and publishing but the result is not picked up in:

https://github.com/PX4/Firmware/blob/7996ee496c5bcc0d4a5e856495a677d8455f2748/src/modules/mavlink/mavlink_messages.cpp#L4431

The update was `false`, hence not updated.

I assume this is the culprit, right @dagar?

https://github.com/PX4/Firmware/blob/7996ee496c5bcc0d4a5e856495a677d8455f2748/src/modules/uORB/Subscription.cpp#L101

If I change `(_last_generation % 100 == 0)` to `(_last_generation % 1 == 0)` it works :smile:.

julianoes, 2019-11-13:
This might be related to https://github.com/mavlink/MAVSDK/issues/885.

dagar, 2019-11-13:
I suppose that's possible with mavlink starting before the land detector. Let's see if that's actually happening here.

The `uORB::Subscription` optimization was intended to be temporary and replaced with something like a unique identifier per orb topic (index) and a bitset. Basically something that allows for an efficient `orb_exists` equivalent.

julianoes, 2019-11-13:
Is this optimization really needed? It seems to bite us quite a bit.

dagar, 2019-11-13:
> Is this optimization really needed? It seems to bite us quite a bit.

Yes, every module is checking for subscription updates every iteration. We need some mechanism to do this cheaply for topics that don't exist and might never exist. Actual `orb_exists` calls are pretty expensive. 

[image]

Try lowering the retry from 100 to 10 and see how it impacts cpu usage (push to a pr-* branch and let the test rack run it).

https://github.com/PX4/Firmware/blob/7996ee496c5bcc0d4a5e856495a677d8455f2748/src/modules/uORB/Subscription.cpp#L101

I'm also thinking about the mavlink startup. For a number of reasons it would probably make sense to start mavlink last, but one major reason to start it earlier would be to get error messages out to the user.

Another idea, update `mavlink boot_complete` so that it does a pass through `List<MavlinkOrbSubscription *> _subscriptions` to force subscribe one last time.

dagar, 2019-11-14:
It should be publishing continuously at 1 Hz (or immediately on any change).

https://github.com/PX4/Firmware/blob/6369ae858a79cc410c1963c1c0046bfd8a59f2a7/src/modules/land_detector/LandDetector.cpp#L85-L86

Is that not the case at boot with lockstep?

julianoes, 2019-11-15:
Timestamps should behave similarly in lockstep. I'll check that!

julianoes, 2019-11-15:
Now I can't reproduce it anymore :see_no_evil:.

julianoes, 2019-11-15:
Ok it happened again.

I think you're right that it's actually related to lockstep. What's happening is that MAVLink is already running even if Gazebo is not running yet, hence time is still at 0. The land detector will start publishing when sensor input and time starts flowing, however, by that time mavlink_messages.cpp has already tried many times.

Should we wait to run the mavlink loops as long as time is not rolling yet?

dagar, 2019-11-15:
> Should we wait to run the mavlink loops as long as time is not rolling yet?

Can we have the `simulator start` block waiting for actual data?

https://github.com/PX4/Firmware/blob/11b157396192ca535bf721145861bdaed8185f34/ROMFS/px4fmu_common/init.d-posix/rcS#L208

julianoes, 2019-11-15:
Ah, that might be the easiest solution! Thanks, I'll try that.

julianoes, 2019-11-18:
Ok, I've changed it in https://github.com/PX4/Firmware/pull/13075/commits/f913cc24aec47fab3c55aad35841d41e30a4199d.

Closing.

closed 2019-11-18 by julianoes