PX4/PX4-Autopilot #13465: Herelink occasionally fails to load parameters

SwiftGust, 2019-11-14:
**Describe the bug**
A clear and concise description of the bug.

I'm not sure it is Herelink specific issue.

When running 3 mavlink instances on pixhawk4,
Occasionally Herelink fails to load PX4 parameters, hangs on loading parameters
but other features such as RC control(joystick), start mission through Herelink still available.
This never resolved rebooting, power-cycling the vehicle & Herelink

From message seen from QGC from laptop relayed from onboard computer through mavros,
messages are keep displayed unknown parameter:#(parameter ids)

System setup
Instance / Baud rate / Mode /  Rate / Terminal / Description 
Instance 0 / 57600 /  Normal / 0 / Herelink / Having issue with loading parameters occasionally
Instance 1 / 921600 / Onboard / 800000 / Onboard Computer / using mavros to connect QGC to the laptop, never had issue with parameter loading
Instance 2 / 115200 / Normal / 0 / Mavlink Gimbal / camera mount

**To Reproduce**
Steps to reproduce the behavior:
1. Random, unknown cause yet, hard to reproduce on our side.

**Expected behavior**
A clear and concise description of what you expected to happen.

Herelink cannot load PX4 parameters while other functionalities such as joystick control, mission start, take-off, etc... are working

julianoes, 2019-11-15:
Do the params appear when you press "refresh" in QGC?

SwiftGust, 2019-11-18:
Nope, cannot refresh since QGC does not appear related interfaces due to parameter reading failed on herelink side
This sometimes solved by complete power-cycle, but just rebooting doesn't solve anything.

SwiftGust, 2019-11-28:
This issue could be able to reproduce when QGC connected to the terminal through serial radio link
then the trying to connect QGC from onboard PC through laptop terminal or vice versa, only single QGC can read parameters.

It seems firmware does not respond to parameter request from concurrent multiple QGCs
and also, the firmware seems to reject RTCM message injection from other GCS than main GCS.