PX4/PX4-Autopilot #13466: Rover: Mission with landing waypoint rejected

jbeyerstedt, 2019-11-14:
A mission, that contains/ ends with a landing waypoint gets rejected:
```
WARN  [navigator] Mission rejected: waypoint within landing flare.
ERROR [navigator] mission check failed
```

Mission Plan: [rover-landing.plan.zip](https://github.com/PX4/Firmware/files/3845650/rover-landing.plan.zip)  
Tested in headless gazebo SITL simulation (`HEADLESS=1 make px4_sitl_default gazebo_rover`) with the current master.

julianoes, 2020-02-14:
@jbeyerstedt hm, and why do you add a landing for a rover at the end of the mission?

jbeyerstedt, 2020-02-14:
I’m not sure any more, why I created a mission in that way. Maybe because I just replicated what I did for a drone mission. Or I tried this because I had issues with the gazebo rover not stopping at a waypoint because the drag was too low in the simulation.

Just finishing the mission with a normal waypoint would be sufficient, right?

Thinking about the issue a second time, it might just be that the error message is not quite accurate for a rover.

julianoes, 2020-02-14:
> Just finishing the mission with a normal waypoint would be sufficient, right?

Yes, that should work.

> Thinking about the issue a second time, it might just be that the error message is not quite accurate for a rover.

I suppose not, no :).

I suggest we close this until you have a suggestion what exactly should be changed.

julianoes, 2020-02-14:
Somewhat connected: #13769

jbeyerstedt, 2020-02-14:
I think closing this is ok for now. I don’t know how much effort it is to change the message for a rover. And in the end it’s not a big issue, because with a real rover it should work without a landing waypoint. So just close this.

(I’m currently on vacation without any laptop so I can’t take a deeper look at it again for another week.)

closed 2020-02-14 by julianoes