PX4/PX4-Autopilot #13471: v1.10.0-rc quadcopter doesn't take off in HITL/jMAVsim

langrind, 2019-11-14:
**Describe the bug**
Set up HITL simulation with jMAVsim on Ubuntu 18.04,  Pixhawk 4 with v1.10.0-rc, locally built px4_fmu-v5-multicopter. 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the recipe from [px4 website](https://dev.px4.io/v1.9.0/en/simulation/hitl.html#jmavsimgazebo-hitl-environment)
2. Using Manual/Stabilized mode, arm vehicle and apply throttle
3. Vehicle does not take off
4. Mavlink shell shows actuator_outputs_1 with *valid* values and actuator_outputs_0 with 0 values
5. MAVLink inspector in QGC shows no HIL_ACTUATOR_CONTROLS, and SERVO_OUTPUT_RAW values are all 0

**Expected behavior**
SERVO_OUTPUT_RAW or HIL_ACTUATOR_CONTROLS should follow actuator_outputs_1 and the vehicle should take off in simulation

**Log Files and Screenshots**
[Flight Review Link](https://review.px4.io/plot_app?log=8890b998-3b24-4a96-af5a-da4c10910bff)

Add screenshots to help explain your problem.

**Drone:**
- Pixhawk4 running in HITL Quad Mode
- Photo of the IMU / autopilot setup if possible.
[image]

julianoes, 2019-11-15:
That's not good but thanks for the detailed report!

langrind, 2019-11-15:
Originally I stated : "MAVLink inspector in QGC shows no HIL_ACTUATOR_CONTROLS, and SERVO_OUTPUT_RAW values are all 0".

But this is misleading. The MAVLink Inspector in QGC can only show the messages forwarded by jMAVSim. What's more interesting is:

1) Once the problem is happening, kill the simulator
2) Connect QGC to the Pixhawk via USB
3) Use the MAVLink inspector

This shows:

* HIL_ACTUATOR_CONTROLS _is_ generated by the Pixhawk, but it is all zeros
* SERVO_OUTPUT_RAW flips back and forth between all-zeros sometimes and correct (or at least sensible) values other times

It seems to me that the publication of uOrbs actuator_outputs instance 0 and instance 1 is confusing the mavlink module's generation of HIL_ACTUATOR_CONTROLS and SERVO_OUTPUT_RAW. 

```
listener actuator_outputs

TOPIC: actuator_outputs 2 instances

Instance 0:
 actuator_outputs_s
	timestamp: 712451977  (0.010010 seconds ago)
	noutputs: 8
	output: [0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000, 0.0000]

Instance 1:
 actuator_outputs_s
	timestamp: 712226919  (0.236525 seconds ago)
	noutputs: 12
	output: [1358.0000, 1408.0000, 1515.0000, 1323.0000, 1500.0000, 1500.0000, 900.0000, 900.0000, 1500.0000, 1500.0000, 1500.0000, 1500.0000, 0.0000, 0.0000, 0.0000, 0.0000]
```

If only instance 0 was published, and it had the correct values, I bet it would work.

Also, I tested a Pixhawk 2.1 (Cube) running px4_fmu-v3_default, and the behavior was 100% identical.

langrind, 2019-11-15:
This change causes the HIL_ACTUATOR_CONTROLS mavlink message to have non-zero values, and now the drone takes off in jMAVSim and I can fly around Zurich crashing into virtual trees:

```
langrind@nik:~/Firmware$ git diff
diff --git a/src/modules/mavlink/mavlink_messages.cpp b/src/modules/mavlink/mavlink_messages.cpp
index 61a55b8b22..4e258c6165 100644
--- a/src/modules/mavlink/mavlink_messages.cpp
+++ b/src/modules/mavlink/mavlink_messages.cpp
@@ -3065,7 +3065,7 @@ private:
 protected:
        explicit MavlinkStreamHILActuatorControls(Mavlink *mavlink) : MavlinkStream(mavlink),
                _status_sub(_mavlink->add_orb_subscription(ORB_ID(vehicle_status))),
-               _act_sub(_mavlink->add_orb_subscription(ORB_ID(actuator_outputs))),
+              _act_sub(_mavlink->add_orb_subscription(ORB_ID(actuator_outputs), 1)),
                _act_time(0)
```

I wonder if this is the correct fix?

langrind, 2019-11-16:
https://github.com/PX4/Firmware/pull/13488 is my proposed fix

commit ba5efa56 (2019-11-17): px4io: When running HITL, don't publish actuator_outputs. Fixes #13471.

When running in HITL mode, pwm_out_sim publishes actuator_outputs.

px4io unconditionally publishes the uOrb actuator_outputs. When HITL
is configured, the px4io copy of the uOrb has all zeros.

The result is that there are two publications, one valid, and one
all-zeros. This causes the HIL_ACTUATOR_CONTROLS mavlink message
to be incorrect (all-zeros) and the SERVO_OUTPUTS_RAW mavlink
message to be inconsistent, as it takes the data from one or the
other uOrb randomly each cycle.

The fix is to let px4io know that HITL is [... cut]
commit 4e126c06 (2019-11-20): px4io: When running HITL, don't publish actuator_outputs. Fixes #13471.

When running in HITL mode, pwm_out_sim publishes actuator_outputs.

px4io unconditionally publishes the uOrb actuator_outputs. When HITL
is configured, the px4io copy of the uOrb has all zeros.

The result is that there are two publications, one valid, and one
all-zeros. This causes the HIL_ACTUATOR_CONTROLS mavlink message
to be incorrect (all-zeros) and the SERVO_OUTPUTS_RAW mavlink
message to be inconsistent, as it takes the data from one or the
other uOrb randomly each cycle.

The fix is to let px4io know that HITL is [... cut]
closed 2019-11-17 by LorenzMeier