PX4/PX4-Autopilot #13472: Firmware version is weird number

tuloski, 2019-11-14:
When I run the simulation with gazebo and mavros (using IRIS model), on QGC I see a weird version for the firmware (something like 48.3.4). Where is this number coming from and why is it there? Should't it be related to the PX4/Firmware number, so something like 1.9.2?

dagar, 2019-11-14:
Make sure you have all the tags locally. 

`git fetch --all --tags`

tuloski, 2019-11-18:
I'll try that, but in which repo? Firmware or sitl?
But just for knowledge where is that number coming from? Mavlink from the autopilot?

dagar, 2019-11-18:
In `Firmware`. It's set at build time based on the git tags you have locally (git tag -l).

tuloski, 2019-11-18:
Ok, but how does QGC get it?

dagar, 2019-11-18:
Over mavlink from firmware. https://mavlink.io/en/messages/common.html#AUTOPILOT_VERSION

closed 2020-03-02 by julianoes