PX4/PX4-Autopilot #13479: Unexpected Freefall from Loiter Mode

pifordi, 2019-11-15:
I have flight three times with my vtol plane two days ago. In the last flight,  during loiter mode at 280meters, the plane start to descent its altitude. I commanded RTL and it return back to home but It still losing altitude during rtl mode. I switch to manual at about 10 meters and landed the plane. Loiter mode does not appear in flight log records. I think that fw control module stops to work in air. The plane did not try to reach cruise airspeed during this period. Actually, it did not try to correct anything. 
After landing, I was able to run lifter motors, servos and pusher motor. The plane was able to take off and landing normally after landing. There was no hardware problem. You can see easily, there is long dropout on logs at all plots.

Flight log: https://review.px4.io/plot_app?log=ecc40545-1384-4185-828e-c14322655122

julianoes, 2019-11-15:
What I can see is that you had RC loss for a while right in the beginning. Did you notice that?

pifordi, 2019-11-15:
The plane always fly in mission mode. Do not matter manual-control lost warnings for me.

julianoes, 2020-02-14:
> The plane always fly in mission mode. Do not matter manual-control lost warnings for me.

You're right. `NAV_RCL_ACT` is set to `0`.

@Antiheavy maybe you have some insights here?

sfuhrer, 2021-01-18:
There was really something very weird going on, never seen something similar. Looks like the whole controller stops.
[image]
Please re-open/comment if you have seen that again and thus want to keep it open - if it was a software issue, the chance is high though it has been fixed since.

closed 2021-01-18 by sfuhrer