PX4/PX4-Autopilot #13480: [FW][gazebo]Fixed wing can't land to home point when I choose RTL mode

xdwgood, 2019-11-15:
I can't land in gazebo after I use the latest master branch to set the `RTL_TYPE = 0 `and `RTL_LAND_DELAY = 0.5` parameters.

sitl log : 

What parameters can be set, when I choose RTL mode, the fixed wing can automatically land to the ground.

xdwgood, 2019-11-15:
@RomanBapst  Do you know the correct parameter configuration? Thank you

Antiheavy, 2019-11-15:
RTL_TYPE =1 is typically the best option for fixed wing. It uses the mission landing pattern to RTL.

xdwgood, 2019-11-16:
@Antiheavy  thanks your replay
I also try set RTL_TYPE =1, but still not land when I switch RTL mode.

How should I set the mission land?

Antiheavy, 2019-11-16:
Huh. Did you upload a mission with a landing pattern?  See https://docs.px4.io/master/en/flight_modes/return.html

xdwgood, 2019-11-16:
@Antiheavy  FW can land when I choose mission mode and add a landing point.
just like this: 
[image]

But when I choose RTL mode, FW will not land but hover.
Sorry, I still don't understand what I should do, I just want to get an RTL similar to the rotor (automatic landing to the takeoff point).

Antiheavy, 2019-11-16:
Are you simulating a fixed wing or a VTOL aircraft?  I don’t understand how a forced wing could hover. There must be something strange with your setup, I just can’t tell what.

xdwgood, 2019-11-16:
> Are you simulating a fixed wing or a VTOL aircraft? 

FW

> There must be something strange with your setup,

maybe,But I have not changed other parameters.

Anyway, thank you again for your reply.

Antiheavy, 2019-11-16:
@xdwgood I notice that you are using the very latest daily build versions of QGC.  I know there are a lot of changes happening right now with the mission planning section.  Have you tried this with the official released version?  Maybe there is a bug with the landing pattern in the daily builds?
http://qgroundcontrol.com/downloads/

xdwgood, 2019-11-17:
@Antiheavy  I tried to use the released version of qgc. After executing rtl, the drone will not land, and do circular motion at 100 meters.

Antiheavy, 2019-11-17:
> and do circular motion at 100 meters.

does it do this around the Home location or somewhere else?

xdwgood, 2019-11-17:
> does it do this around the Home location or somewhere else?

@Antiheavy  it around the Home location

Antiheavy, 2019-11-17:
do you get errors when trying to upload the mission to the autopilot?

xdwgood, 2019-11-17:
I have not updated the mission. My operation flow is: commander take off--> switch RTL mode

xdwgood, 2019-11-17:
Or currently px4 does not support automatic landing in rtl mode, we have to upload a mission and then execute rtl?

Antiheavy, 2019-11-17:
huh... i'm unfamiliar with these new log messages:
[image]

I wonder if something is broken with the new RTL stuff around safe points.  @sfuhrer can you take a look at this?

xdwgood, 2019-11-17:
@sfuhrer   @Antiheavy 
My operation flow: start gazebo FW simulation --> upload a FW landing point ---> send take off command ---> switch mode to RTL

log link：  https://logs.px4.io/plot_app?log=5c633144-1470-4bd1-af18-68d189d75a7e

[image]

In the log print information, the landing has been prompted, but in the simulation, the drone has been flying forward without landing.

[image]

sfuhrer, 2019-11-17:
Hi @xdwgood,

I've just tried it with the same setup as you described, and it worked for me as expected (landing at mission landing).

https://review.px4.io/plot_app?log=693a3338-5cb1-4059-9f0e-48a261ff8146
https://review.px4.io/plot_app?log=c8686c3f-23c4-4090-bfb0-b62cc32df9c4

Looking at the logged messages of your flight I can only think of that the mission (in your case landing point) upload failed for some reason. It looks like it has no mission landing, and thus chooses Home to go to in the RTL. Ofc it doesn't land there, but circles. I'm wondering though why it then says "Landing at destination" - it should instead should say "RTL: completed, loitering". I just checked, that's what you get when you RTL in a flight without any mission planned.

Did you check if you get any mission feasibility errors while uploading or when starting the flight? And did you also try with a complete mission, meaning with takeoff, normal wp and then the mission landing?

Antiheavy, 2019-11-17:
If you were using RTL_TYPE 1 it should force you to use the mission landing in all except certain cases like you deleted the mission mid-flight or something. 

Another weird thing is I’d expect to see a logged message that says something like “unable to use mission landing, returning to Home”

Also, the weird thing is that you should not even be able to takeoff in an auto mode if the vehicle does not have a valid mission.

sfuhrer, 2019-11-18:
> Also, the weird thing is that you should not even be able to takeoff in an auto mode if the vehicle does not have a valid mission

That was the case, I had to trigger takeoff over the command line as the mission was not valid (only had a landing pattern but no takeoff). What was your exact procedure @xdwgood ?

xdwgood, 2019-11-19:
> That was the case, I had to trigger takeoff over the command line as the mission was not valid (only had a landing pattern but no takeoff). What was your exact procedure @xdwgood ?

@sfuhrer   I send the takeoff through the command line:
`Commander takeoff`

> And did you also try with a complete mission, meaning with takeoff, normal wp and then the mission landing?

yes，I have tried a mission(wp) and will land automatically when the drone is executed.

Clarify my detailed operation again:  start gazebo FW simulation -->open qgc---> upload a FW landing point ---> send take off command by `commander takeoff` ---> switch mode to RTL--->FW can't land

@Antiheavy @sfuhrer  Thanks Reply

sfuhrer, 2019-11-19:
@xdwgood this was the exact same procedure I used in [this](https://review.px4.io/plot_app?log=693a3338-5cb1-4059-9f0e-48a261ff8146) flight. In your [first log](https://logs.px4.io/plot_app?log=e409a27b-aa70-4ddd-a7b0-f3118a4e6a41) you seem to have RTL_TYPE=0, in which case it is normal that it will go to home and loiter there. In your [second log](https://logs.px4.io/plot_app?log=5c633144-1470-4bd1-af18-68d189d75a7e) though, with RTL_TYPE=1, the logged messages indicate that it actually did go to the mission landing. Was this not the case?

xdwgood, 2019-11-19:
>  the logged messages indicate that it actually did go to the mission landing. Was this not the case?

But the FW did not land, but continued to fly straight.

sfuhrer, 2019-11-19:
@xdwgood could you maybe try again? To see if it reproducible. As I couldn't reproduce it so far.

xdwgood, 2019-11-20:
@Antiheavy @sfuhrer 
I updated the code and tried again, everything worked fine, FW landed on the ground. I am sorry to bother you, thank you for your reply, thank you.

closed 2019-11-20 by xdwgood