PX4/PX4-Autopilot #13483: Durandal: manual logger start fails

mhkabir, 2019-11-15:
Starting logger on Durandal (`stm32h7_support` branch) manually from Mavlink console seems to fail:

[image]

It does report that a new log file was opened, after `logger on`:
[image]

However these files are always empty:

[image]

dagar, 2019-11-15:
It seems to be working on the test rack. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-hardware/detail/stm32h7_support/26/pipeline/661#step-840-log-5

[image]

[image]

[image]

davids5, 2019-11-15:
@mhkabir - I think there may be a power sequencing issue with the SD.
What does dmesg show?

LorenzMeier, 2019-11-15:
@mhkabir How do you power the unit? We have flown successfully with valid logs. I presume you did start that on the bench?

@davids5 I have confirmed the power sequencing issue. Steps:

- Remove the SD card
- Unpower Durandal
- Plug USB in
- Plug the SD card in BEFORE the boot loader lights go out
- run ```logger stop && logger start -e```

LorenzMeier, 2019-11-15:
I would leave this issue open so we can add an errata to the documentation @hamishwillee as a known hardware limitation if powered via USB. But it is not a software issue per se (and maybe we can work around it in SW later).

mhkabir, 2019-11-15:
Some more details:
- I'm powering via USB only.
- Logging works if it's starting from boot.
-

mhkabir, 2019-11-15:
@LorenzMeier yes, this is on bench, powered from USB only.

- I'm powering via USB only.
- Logging *does work* if it's started from boot. (SDLOG_MODE)
- I didn't touch the SD card.

mhkabir, 2019-11-15:
@davids5 
[image]

zhaoxiaowei1013, 2020-04-17:
have a plan to solve it？I also encountered the same problem。

davids5, 2020-04-17:
@zhaoxiaowei1013 This was addressed in https://github.com/PX4/Firmware/pull/14592 

Please test current master and reopen this issues if it did not solve it.

closed 2020-04-17 by davids5

jareiko, 2020-04-21:
I'm still sometimes getting what might be this problem on master, running off battery power.

At some point the SD card seems to fail, and then nothing works anymore (logging or sd_bench, for example).

[image]

dagar, 2020-04-21:
@jareiko could you open a new issue for that and start collecting any relevant data?

Is that console screenshot showing anything other than logger successfully starting?

Have you noticed any patterns? Does it fail after a certain period of time? Are you getting any potentially relevant error messages from logger or the system? You could check `dmesg` output.

jareiko, 2020-04-21:
That is just a dmesg log, as requested earlier in the thread. Yes, I'll start a new issue for this when I'm looking at it again. For now we're exploring log streaming instead.

davids5, 2020-04-21:
@jareiko - Also have you tried another card?

jareiko, 2020-04-21:
Yes, three different cards, all of which work fine on a Pixhawk 4.

zhaoxiaowei1013, 2020-04-23:
This problem still exists. I tested this problem in the latest master branch
[image]
And all in my  cuav v5+ is work good.

davids5, 2020-06-29:
Resolved by https://github.com/PX4/NuttX/pull/106