PX4/PX4-Autopilot #13486: Make for fmu v2 results in " fatal error: sys/cdefs.h: No such file or directory"

iwishiwasaneagle, 2019-11-15:
**Describe the bug**
Running `make px4_fmu-v2_default` results in multiple error messages pointing to errors with `glibc`

**Expected behavior**
`make` runs without fault and builds px4 for fmu v2

**Log Files and Screenshots**
```
$ make px4_fmu-v2_default
[  0%] Built target git_nuttx_apps
[  0%] Built target nuttx_defconfig_target
[  1%] Built target git_nuttx
[  1%] Built target mixer_gen_6dof
[  1%] Built target uorb_headers
[  1%] Built target mixer_gen
[  1%] Built target ver_gen
[  1%] Built target git_gps_devices
[  1%] Built target git_ecl
[  1%] Built target git_mavlink_v2
[  1%] Built target px4_config_file_target
[  1%] Built target copy_apps_files_target
[  1%] Built target airframes_xml
[  2%] Built target parameters_xml
[  2%] Performing build step for 'px4io_firmware'
[  1%] Built target git_nuttx_apps
[  1%] Built target nuttx_defconfig_target
[  1%] Built target git_nuttx
[  1%] Built target uorb_headers
[  1%] Built target mixer_gen_6dof
[  3%] Built target mixer_gen
[  3%] Built target nuttx_builtin_list_target
[  3%] Built target copy_apps_files_target
[  3%] Built target airframes_xml
[  5%] Built target parameters_xml
[  2%] Built target copy_nuttx_files_target
[  5%] Built target copy_nuttx_files_target
[  2%] Built target nuttx_copy_and_apps_target
[  3%] Built target nuttx_copy_config_dir_target
[  3%] Built target nuttx_config_kconfig_target
[  3%] Built target nuttx_config_target
[  3%] Built target nuttx_context
[  3%] Building CXX object src/lib/perf/CMakeFiles/perf.dir/perf_counter.cpp.obj
[  3%] Built target df_driver_framework
[  3%] Building CXX object src/lib/parameters/tinybson/CMakeFiles/tinybson.dir/tinybson.cpp.obj
[01m[K/opt/drone/PX4/src/lib/perf/perf_counter.cpp:43:10:[m[K [01;31m[Kfatal error: [m[Ksys/queue.h: No such file or directory
   43 | #include [01;31m[K<sys/queue.h>[m[K
      |          [01;31m[K^~~~~~~~~~~~~[m[K
compilation terminated.
make[3]: *** [src/lib/perf/CMakeFiles/perf.dir/build.make:63: src/lib/perf/CMakeFiles/perf.dir/perf_counter.cpp.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:3842: src/lib/perf/CMakeFiles/perf.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
In file included from [01m[K/opt/drone/PX4/platforms/common/include/px4_platform_common/defines.h:42[m[K,
                 from [01m[K/opt/drone/PX4/platforms/common/include/px4_platform_common/posix.h:42[m[K,
                 from [01m[K/opt/drone/PX4/src/lib/parameters/tinybson/tinybson.cpp:40[m[K:
[01m[K/opt/drone/PX4/platforms/common/include/px4_platform_common/log.h:120:10:[m[K [01;31m[Kfatal error: [m[Ksys/cdefs.h: No such file or directory
  120 | #include [01;31m[K<sys/cdefs.h>[m[K
      |          [01;31m[K^~~~~~~~~~~~~[m[K
compilation terminated.
make[3]: *** [src/lib/parameters/tinybson/CMakeFiles/tinybson.dir/build.make:63: src/lib/parameters/tinybson/CMakeFiles/tinybson.dir/tinybson.cpp.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:8348: src/lib/parameters/tinybson/CMakeFiles/tinybson.dir/all] Error 2
[  7%] Built target nuttx_copy_and_apps_target
[  7%] Built target nuttx_config_kconfig_target
[  7%] Built target nuttx_copy_config_dir_target
[  9%] Built target nuttx_config_target
[  3%] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/actuator_armed.cpp.obj
[  3%] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/actuator_controls.cpp.obj
[  4%] Building CXX object msg/CMakeFiles/uorb_msgs.dir/topics_sources/actuator_direct.cpp.obj
[  9%] Built target nuttx_context
In file included from [01m[K/opt/drone/PX4/build/px4_fmu-v2_default/msg/topics_sources/actuator_armed.cpp:38[m[K:
[01m[K/opt/drone/PX4/platforms/common/include/px4_platform_common/log.h:120:10:[m[K [01;31m[Kfatal error: [m[Ksys/cdefs.h: No such file or directory
[... 86 lines cut ...]
  120 | #include [01;31m[K<sys/cdefs.h>[m[K
      |          [01;31m[K^~~~~~~~~~~~~[m[K
compilation terminated.
make[6]: *** [msg/CMakeFiles/uorb_msgs.dir/build.make:575: msg/CMakeFiles/uorb_msgs.dir/topics_sources/actuator_direct.cpp.obj] Error 1
make[5]: *** [CMakeFiles/Makefile2:444: msg/CMakeFiles/uorb_msgs.dir/all] Error 2
make[4]: *** [Makefile:106: all] Error 2
make[3]: *** [src/drivers/px4io/CMakeFiles/px4io_firmware.dir/build.make:112: external/Stamp/px4io_firmware/px4io_firmware-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:5732: src/drivers/px4io/CMakeFiles/px4io_firmware.dir/all] Error 2
make[1]: *** [Makefile:106: all] Error 2
make: *** [Makefile:195: px4_fmu-v2_default] Error 2
```

[image]

```
$pkgfile /usr/include/sys/cdefs.h
core/glibc
```
```
$pkgfile /usr/include/sys/queue.h
core/glibc
```

**Additional context**
I am running 5.3.8-3-MANJARO 64bit

LorenzMeier, 2019-11-15:
Did you check the code out via GIT?

iwishiwasaneagle, 2019-11-15:
> Did you check the code out via GIT?

yes

julianoes, 2019-11-19:
@iwishiwasaneagle are you using ninja? And what shell are you using?

And can you do `make clean` and then paste the full output please?

iwishiwasaneagle, 2019-11-19:
> @iwishiwasaneagle are you using ninja? 
No

>And what shell are you using?
ZSH, but this issue presists when I use bash too.
 
> And can you do `make clean` and then paste the full output please?

No output at all from `make clean`. Same error when I run `make px4_fmu-v2_default` afterwards

julianoes, 2019-11-19:
I don't think that's the full output.

For me for instance I also see this:
```
make px4_fmu-v2_default                                                 
-- PX4 version: v1.10.0-beta4-273-gbbc028a9e9
-- PX4 config file: /home/julianoes/src/Firmware/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/julianoes/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /home/julianoes/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/julianoes/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /home/julianoes/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/julianoes/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "3.7.5") 
-- Found PY_jinja2: /home/julianoes/.local/lib/python3.7/site-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-88-gd76b704
-- Building and including px4_io-v2_default
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: /home/julianoes/src/Firmware/../cmsis-svd/data/STMicro/STM32F427.svd
-- Configuring done
-- Generating done
```

iwishiwasaneagle, 2019-11-19:
Im running ` make px4_fmu-v2_default 2>&1 | xclip -sel clip` to copy the output

julianoes, 2019-11-19:
> -- The CXX compiler identification is GNU 9.2.0
-- The C compiler identification is GNU 9.2.0

We currently use GCC 7.2.1 for STM32.

Check:

https://github.com/PX4/Firmware/blob/01c7a475ff36a4c1df9b17ccbdae62b5967632cf/Tools/setup/arch.sh#L90-L111

iwishiwasaneagle, 2019-11-19:
> > -- The CXX compiler identification is GNU 9.2.0
> > -- The C compiler identification is GNU 9.2.0
> 
> We currently use GCC 7.2.1 for STM32.
> 
> Check:
> 
> https://github.com/PX4/Firmware/blob/01c7a475ff36a4c1df9b17ccbdae62b5967632cf/Tools/setup/arch.sh#L90-L111

Okay that's definitely changed the error haha thanks!

```
-- PX4 version: v1.10.0-beta4-158-ge3964d5150
-- PX4 config file: /opt/drone/PX4/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- PX4 lockstep: disabled
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "3.7.4") 
-- Found PY_jinja2: /usr/lib/python3.7/site-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-88-gd76b704
-- Building and including px4_io-v2_default
-- ROMFS: px4fmu_common
-- ROMFS: Adding rc.board_defaults
-- ROMFS: Adding rc.board_sensors
-- ROMFS: Adding rc.board_mavlink
-- Found SVD: 
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/drone/PX4/build/px4_fmu-v2_default
[0/721] git submodule platforms/nuttx/NuttX/nuttx
[1/721] Creating airframes.xml
[2/721] git submodule platforms/nuttx/NuttX/apps
[3/721] Generating serial_params.c
[4/721] git submodule src/lib/ecl
[8/721] Creating directories for 'px4io_firmware'
[9/721] git submodule src/drivers/gps/devices
[16/721] Copying NuttX config nsh
[17/721] git submodule mavlink/include/mavlink/v2.0
[18/721] ROMFS: copying, generating airframes
[20/721] Generating git version header
[21/721] Generating parameters.xml
[22/721] Generating uORB topic headers
[23/721] Copying NuttX compressed config nsh and inflating
[24/721] Generating nuttx/include/nuttx/version.h, nuttx/include/nuttx/config.h
[25/721] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DriverFramework_NuttX.cpp.obj
[26/721] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DevMgr_Nuttx.cpp.obj
[27/721] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/DFList.cpp.obj
[28/721] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/SyncObj.cpp.obj
[29/721] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_controller.cpp.obj
[30/721] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_pitch_controller.cpp.obj
[31/721] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_roll_controller.cpp.obj
[32/721] Building CXX object src/lib/DriverFramework/framework/src/CMakeFiles/df_driver_framework.dir/Time.cpp.obj
[33/721] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_wheel_controller.cpp.obj
[34/721] Building CXX object src/lib/ecl/attitude_fw/CMakeFiles/ecl_attitude_fw.dir/ecl_yaw_controller.cpp.obj
[35/721] Building CXX object src/lib/ecl/geo/CMakeFiles/ecl_geo.dir/geo.cpp.obj
[36/721] Building CXX object src/lib/ecl/geo_lookup/CMakeFiles/ecl_geo_lookup.dir/geo_mag_declination.cpp.obj
[... 884 lines cut ...]
[197/207] Linking CXX static library platforms/nuttx/src/px4/stm/stm32f1/io_pins/libarch_io_pins.a
[198/207] Linking CXX static library src/modules/px4iofirmware/libpx4iofirmware.a
[199/207] Generating nuttx/drivers/libdrivers.a
[200/207] Generating nuttx/fs/libfs.a
[201/207] Generating nuttx/arch/arm/src/libarch.a
[202/207] Generating nuttx/sched/libsched.a
[203/207] Generating nuttx/libs/libc/libc.a
[204/207] Building C object platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.c.obj
[205/207] Linking CXX executable px4_io-v2_default.elf
FAILED: px4_io-v2_default.elf 
: && /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++  -mcpu=cortex-m3 -mthumb -Os -DNDEBUG  --specs=nosys.specs -fuse-ld=gold platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.c.obj  -o px4_io-v2_default.elf -L/opt/drone/PX4/build/px4_fmu-v2_default/external/Build/px4io_firmware/external/Install/lib -nostartfiles -nodefaultlibs -nostdlib -nostdinc++ -fno-exceptions -fno-rtti -Wl,--script=/opt/drone/PX4/build/px4_fmu-v2_default/external/Build/px4io_firmware/NuttX/nuttx-config/scripts/script.ld -Wl,-Map=px4_io-v2_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libs/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libs/libxx/libxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc src/modules/px4iofirmware/libpx4iofirmware.a NuttX/apps/libapps.a platforms/nuttx/src/px4/stm/stm32f1/io_pins/libarch_io_pins.a src/lib/mixer/libmixer.a src/lib/rc/librc.a src/lib/perf/libperf.a src/lib/output_limit/liboutput_limit.a NuttX/nuttx/arch/arm/src/libarch.a boards/px4/io-v2/src/libdrivers_board.a platforms/nuttx/ [... cut ...]
/usr/bin/arm-none-eabi-ld.gold: error: section .ARM.exidx overflows end of region flash
/usr/bin/arm-none-eabi-ld.gold: error: section .data overflows end of region flash
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[704/721] Generating nuttx/libs/libc/libc.a
FAILED: external/Stamp/px4io_firmware/px4io_firmware-build 
cd /opt/drone/PX4/build/px4_fmu-v2_default/external/Build/px4io_firmware && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [Makefile:195: px4_fmu-v2_default] Error 1
```

closed 2019-11-20 by julianoes