PX4/PX4-Autopilot #13487: FW HIL config no longer works

apoissant95, 2019-11-15:
**Describe the bug**
I am trying to send offboard position targets to a FW in HIL mode, so I am trying out v1.10. However, HIL mode no longer works with the current master. For manual mode, the RC plane in xplane 10 does not respond to my throttle or roll/pitch/yaw commands via a Taranis Q X7 RC. The RC plane just sits in place. For mission mode, the RC plane again just sits there, does not take off, waits about 15 seconds, and then qgc says the plane completed its mission. The setup worked with stable v1.9.2. 

**To Reproduce**
Follow setup steps for xplane 10 at: https://dev.px4.io/v1.9.0/en/simulation/hitl.html
Steps to reproduce the behavior:
1. Open qgc v3.5.5 and xplane10. 
2. Plug in Pixhawk 4 running v1.10 and HILStar (XPlane) airframe via USB. 
3. Set HIL Config by Widgets --> HIL Config --> Connect. (enable sensor level HIL, use newer actuator format)
4. Change mode to manual mode with RC on. 
5. Try to fly plane. Note the plane in xplane does not move. 
6. Change mode to mission mode with a valid mission uploaded to the Pixhawk. Note the plane does not execute the mission. 
7. Repeat above steps with v1.9.2 flashed. Note nominal behavior for manual and mission modes. 

**Expected behavior**
When manual mode is engaged, the plane in xplane can be controlled with a Taranis Q X7. When mission mode is engaged, the plane executes the uploaded mission. Expected behavior the same as v1.9.2. 

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).
https://logs.px4.io/plot_app?log=78b9472c-5d0b-4eae-958f-4c4b6b9d0cdd
Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone. fixed wing, HILStar (Xplane) airframe
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

julianoes, 2019-11-19:
Thanks for the issue. Is it fixed with latest master?

There have been two fixes just recently:
https://github.com/PX4/Firmware/pull/13488
https://github.com/PX4/Firmware/pull/13504

michael-borkowski, 2019-11-19:
We are experiencing a similar problem with the newest v1.10 firmware. Even though we are not using a plane but a quadcopter in HIL, the result is similar, the quadcopter is not moving (neither in jMAVSim nor in Air Sim).

We were able to trace the problem back to commit 349469cf753d91c6da864d83bd8f3737e20121c7. Using the latest master does resolve the issue.

apoissant95, 2019-11-19:
@julianoes I just uploaded the latest master just now and I am still facing the same problems. However, now qgc rejects any attempt to switch to mission mode even though I have a mission uploaded to the Pixhawk. Additionally, when I try to fly manual it looks like xplane isn't receiving any of the RC inputs because none of the control surfaces move when I try to use my RC.

apoissant95, 2019-11-19:
@julianoes for some reason after reflashing the master it suddenly started working again. Thanks for the help!

closed 2020-02-19 by julianoes

rg-28, 2021-06-19:
Currently I am trying to implement HITL of fixed wing using Gazebo (Plane URDF), PX4-Autpilot version is 1.11.1 and the airframe that I am using is HILSTAR(XPlane). I am trying to send commands in offboard mode and in the QGC it shows mode switch and getting armed but after that nothing is happening even if it is receiving other commands ([setposition](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED)). Also in gazebo the propellers are not rotating on arming but if I switch the mode to stabilize then it starts moving properly.
Any help would be appreciated.