PX4/PX4-Autopilot #13496: vl53lx long range

tech0x, 2019-11-18:
By default distance sensor vl53lxx work in low range mode, with distance only up to 1m, but in long range mode this sensor can work with distances up to 2 meters and rate 10 hz.

Can You add option in source or somewhere else to switch with long range distance measurement ?

julianoes, 2019-11-19:
Can you point us to a datasheet and the source code of the driver. I'm lacking some context.

tech0x, 2019-11-19:
This is datasheet.
https://www.st.com/resource/en/datasheet/vl53l0x.pdf
This is driver 
https://github.com/PX4/Firmware/blob/master/src/drivers/distance_sensor/vl53lxx/vl53lxx.cpp

tech0x, 2019-11-19:
Test distance I make on the arduino with this code
https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino
Long range mode was used.

julianoes, 2019-11-20:
Would you be able to make a pull request which adds a param to configure this?
@mcsauder do you think a param for that would make sense?

tech0x, 2019-11-27:
I try to make this without any success.
I have only arduino working example with long range distance.
This Is may be helpfull ?

dagar, 2019-11-27:
I would actually question the usefulness of the long range mode. It's even more susceptible to interference.

mhkabir, 2019-11-27:
I'm going to close this given that the sensor is terrible in the long ranges mode and we're not aware of anyone using it regularly with the in-tree driver.

@tech0x if you get it working, we are happy to accept PRs!

closed 2019-11-27 by mhkabir

tech0x, 2019-11-27:
I can find cpp working code with longe range it is enough ?