PX4/PX4-Autopilot #13498: Drone rotates towards setpoints in Precision Land flight mode in 1.9.x and 1.10.0

dvornikov-aa, 2019-11-18:
Drone rotates towards setpoints and subsequently fails to land on the target if precision landing starts outside of the mission (by command `commander mode auto:precland`) . The drone yaw was always **0** in **1.8.x**. The worst part is that I can't reproduce this behavior in **SITL** or **HITL**.

**Steps to reproduce the behavior:**

1. Takeoff above **IR-Lock** beacon.
2. Call `commander mode auto:precland`.
3. Watch the precision landing process.

**Expected behavior**
The drone yaw must always be constant to perform a successful precise landing.

**Log Files and Screenshots**
[Log](https://logs.px4.io/plot_app?log=85b20945-deca-4b19-b7d5-7cc8e6bdc2c7)

[image]

**Drone:**
F450-based Quadrotor x

[image]

julianoes, 2020-03-02:
FYI @bresch

julianoes, 2020-06-02:
@dvornikov-aa is this still an issue or have you found a solution or workaround?

kpetrykin, 2020-06-02:
@julianoes yes, the problem is still there. The solution was not found yet.

julianoes, 2020-06-02:
And is it also there in master/1.11?

@dusan19 do you have an idea about this?

dusan19, 2020-06-02:
I see that, precland doesnt set the triplet yaw. 

The triplets were updated only 3 times in this log, each time the yaw was invalid which should result in the flight task calling `_set_heading_from_mode()`. 
So the yaw sp should be towards setpoint until the drone is close horizontally to the position setpoint when the yaw sp should become locked.
Im not sure why this doesnt happen.

I see that the triplet.current.valid is true but `triplet.current.position_valid` is `false`, but the triplet position is finite so the flight task still uses these finite values (checks for `is_finate` rather than the valid flag).
Also `_set_heading_from_mode()` computes the distance to `_target`, but the `_target` is set after the `_set_heading_from_mode()` is called, so `_target` from the previous iteration is used, this might be relevant in this case when the triplet is not updated all the time.

Anyway it would require more detailed debugging, also I never used precland outside of the mission

amefabris, 2024-03-08:
I am facing the same issue on v1.14.0 with a custom drone in simulation (gazebo-classic), but it's really hard to track down the issue. The yaw of the setpoint triplets is always 0, however the drone is changing the yaw and moving towards the target.