PX4/PX4-Autopilot #13506: PX4 SITL max motor speed

waytry, 2019-11-18:
__Describe the question__  
I was trying to use a custom quadcopter model in PX4 SITL. I created a sdf file for the Crazyflie 
 to replace the default iris sdf model. And I successfully launched the PX4 SITL by following [Launching Gazebo with ROS Wrappers](https://dev.px4.io/master/en/simulation/ros_interface.html#launching-gazebo-with-ros-wrappers).  Basically what I did was
```
cd <Firmware_clone>
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 mavros_posix_sitl.launch vehicle:="crazyflie"
```

In the SITL, the PX4 node sends motor speed to the Gazebo simulation (the gazebo_motro_model plugin in particular). As the crazyflie has max motor speed of 3000 rev/s differs from the 1100 rev/s of iris, I wanted to modify this parameter in PX4 firmware. I tried to search "max motor speed" and "max rotor speed". But couldn't find a parameter to specify the max motor speed. Is there a way to modify the max motor speed?

Thanks

julianoes, 2020-03-02:
Did you check in https://github.com/PX4/sitl_gazebo/ ?

julianoes, 2020-06-02:
Closing, there was no response.

closed 2020-06-02 by julianoes

mta32, 2020-09-01:
Heyy!

Can you please send me the sdf model that you have written.I am Having trouble creating my own.
I would appreciate that.

mta32, 2020-09-01:
Heyy!

Can you please send me the crazyflie.sdf file that you had written.I am having trouble creating my own.I would really appreciate that.