PX4/PX4-Autopilot #13507: Thrust value oscillating (repeatedly dropping to 0)

tomek-l, 2019-11-18:
Hey,

My Pixhawk 4 mini hexarotor with PX4 1.9.2 release wasn't maintaining altitude while hovering, so I increased the mid throttle from 50% to 60%.

What I got is this super odd thrust oscillation, while hovering (the thrust value is going to 0 for a short period of time -> see the logs). Even though I'm trying to land, it's only reacting to increasing the throttle, but not to not lowering -> also visible in the logs. I had to use kill switch.

The oscillations occur across all modes.
After going back to 50% mid-throttle the problem disappears. 

The battery warnings are result of the voltage sag (the actual voltage was 3.75V/cell and the default BAT_V_EMPTY is pretty high, at 3.50V/cell). 

Logs:
https://logs.px4.io/plot_app?log=ececa147-2ad7-4aa7-9e0a-bcc0761c06af

tomek-l, 2020-01-29:
@julianoes It really seems that these really large changes in thrust are not correlated to with any sensor input or state estimates. Could you advise how I could get started on investigating/fixing this?

Could it be a repeatedly occurring exception in code that causes the thrust to reset to 0?

steffensol, 2020-02-17:
This seems like it is the land detector acting up again. Multiple issues have been flagged for this, and it seems that the detector is VERY sensitive to hover throttle and minimum throttle values. 

I've used "flightplot" to plot out the ground detection algorithm, see the attached plot showing the detection occurring. 
[image]

This problem has been seen across multiple vehicles, vehicle types, and has been flagged as issues here, with different names and explanations - but they all seem to be linked in my opinion.

#13124 
#11683 
#13231 

I'd say based on your logs that you probably don't need the 60% thrust to hover.
It could very well be the PID values of the position-controller acting up because it seems like it is not following the Z-setpoint at all.

[image]

RomanBapst, 2020-02-17:
> Multiple issues have been flagged for this, and it seems that the detector is VERY sensitive to hover throttle and minimum throttle values.

@steffensol Yes, and that's why @bresch is working on the hover thrust estimator.
https://github.com/PX4/Firmware/pull/13981

steffensol, 2020-02-17:
That's brilliant! 
However, it still doesn't solve the case that the landing detector needs more love&care. In our case, we have the hover thrust set up perfectly, within 1%. Our minimum thrust, however, is only 15% lower than the hover thrust - which causes a "low thrust situation" to trigger very quickly.

RomanBapst, 2020-02-17:
@steffensol What's the reason for it being only 15% lower. Is it a tailsitter :-)
I've personally created a PR request once which does not allow detecting ground contact only by a low thrust situation but people had some concerns. I've seen vehicles crash due to this effect so I do think it's quite serious.

steffensol, 2020-02-17:
The main reason is that we have very large propellers running low RPM, to give high efficiency (We're getting flight times in excess of 1h). Unfortunately this also sometimes leads to loss-of-sync with the ESCs at low throttle. Hover is currently at 40%, and the minimum at 25% - if we go any lower, flight performance is very stuttery.

dagar, 2020-08-02:
MC land detector `ground_contact` detection fixes have been merged in https://github.com/PX4/Firmware/pull/15083.

closed 2020-08-02 by dagar