PX4/PX4-Autopilot #13508: EKF gps hgt timeout - reset to baro

hyonlim, 2019-11-18:
**Describe the bug**
EKF gps hgt timeout - reset to baro problem.
It basically says GPS data is lagged because _gps_hgt_intermittent flag is on.

_gps_hgt_intermittent = !((_time_last_imu - gps_init.time_us) < 2 * GPS_MAX_INTERVAL);

However, it seems very difficult to meet this condition unless otherwise GPS has unstable connection. But it's not. It seems some timing issue.

**To Reproduce**
Not exactly reproducible

**Expected behavior**

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=d8997f75-e796-4a42-9b90-ea5c06a3d54d

**Drone (please complete the following information):**
- FMUv5.

julianoes, 2020-03-02:
FYI @bresch

julianoes, 2020-06-02:
@hyonlim is this still a problem?

bresch, 2020-08-31:
We've never seen that issue since then, closing.

closed 2020-08-31 by bresch