PX4/PX4-Autopilot #13513: RTPS: auto generate yaml file

jlecoeur, 2019-11-19:
The RTPS yaml file has to be manually updated for each new uorb message, which is a bit annoying because 
- This causes CI failures when you forget to update it
- This causes many merge conflicts

Can it be automatically generated?

TSC21, 2019-11-19:
This has been discussed before and the answer is no, because you want to keep consistency on the ID's that are set to each message. The same way you don't automatically generate new ID's for the Mavlink messages.

jlecoeur, 2019-11-19:
What about a unique hash of the message name, or of the message content?

TSC21, 2019-11-19:
> What about a unique hash of the message name, or of the message content?

That was also spoken and an 8-bit hash would result in collisions. And it can't be higher (to include the content), because the ID field has limited size (0-255).

jlecoeur, 2019-11-19:
That is unfortunate. We are already approaching the limit of 150 IDs reserved for non-multitopic messages, are there plans for when that limit is reached?

TSC21, 2019-11-19:
> That is unfortunate. We are already approaching the limit of 150 IDs reserved for non-multitopic messages, are there plans for when that limit is reached?

That limit can be increased. At least for now at development stage we can do that. When the microRTPS starts to be deployed in end systems, then that will be an issue.

jlecoeur, 2019-11-19:
> then that will be an issue.

Sure. MAVLink 1 had the same problem: it was designed with a 255 messages limit, which probably seemed plenty at the time, but was reached since.

We are maybe going further than necessary to solve a mild inconvenience :-). There may be a better solution than a unique hash. But assuming we want a hash, and thus need a larger ID field, is that possible? Is the 8-bit limit imposed by the RTPS protocol, or by PX4's implementation?

TSC21, 2019-11-19:
> We are maybe going further than necessary to solve a mild inconvenience :-). There may be a better solution than a unique hash. But assuming we want a hash, and thus need a larger ID field, is that possible? Is the 8-bit limit imposed by the RTPS protocol, or by PX4's implementation?

As far as I know that's a protocol limitation (specific of the microCDR implementation).

TSC21, 2019-11-19:
@jlecoeur but I don't actually see ourselves requiring more than 256 msgs for internal consumption and onboard communication. A scaleble system must also keep a certain level of consistency and simplicity on the way it exchanges it's data, and have control on what messages are needed and not. Sometimes, it just makes sense to add a specific field to an existent message than just create a new message for the same purpose.

jlecoeur, 2019-11-19:
Absolutely, that brings the question wether all uORB messages should be mapped to RTPS, or just a selected, stable, few.

TSC21, 2019-11-19:
> Absolutely, that brings the question wether all uORB messages should be mapped to RTPS, or just a selected, stable, few.

Not all will be. But we are not at the stage of doing that selection, at least for now.

LorenzMeier, 2019-11-19:
There is a significant difference to MAVLInk in that we’re controlling both sides here (forced by tight integration). We will also need some sort of filter to ensure security and safety of the system for inbound and outbound messages. Otherwise a debut message in ROS can bring the flight controller down via a DDS-like action.

TSC21, 2019-11-19:
> We will also need some sort of filter to ensure security and safety of the system for inbound and outbound messages. Otherwise a debut message in ROS can bring the flight controller down via a DDS-like action.

@LorenzMeier we may explore the security options that Fast-RTPS offers for that same purpose. FastRTPS implements security at three levels: access control of entities, encryption of data and authentication of remote participants.

closed 2020-07-08 by TSC21