PX4/PX4-Autopilot #13514: Help understand inconsistency in altitudes

tuloski, 2019-11-19:
I've been analyzing some logs and I always see this "problem".

[image]

The various altitude measures have some inconsistency IMO.
For example the local_position.z is not at zero when arming (on ground), but it has an offset of around 8 meters. You can see the real altitude from ground in black from the distance_sensor.
Also the vehicle_global_position.alt seems to have the same offset w.r.t. baro data and vehicle_gps_position.alt. The first is at around 57 meters, while the latters at around 49 meters.

Another weird thing is that the "relative alt" I see on QGC matches almost perfectly the distance_sensor, so my questions are:

- Isn't the "relative alt" in QGC taken from local_position.z?
- Is it normal that the local_position.z doesn't reset when arming?
- Why those offsets?

The overall problem of all this, is that during AUTO modes, the altitude references are always wrong and the copter never reaches the desired waypoint because the reference and the altitude used to check the acceptance radius differ of that offset.

julianoes, 2019-11-20:
> Isn't the "relative alt" in QGC taken from local_position.z? If not which data is sent for it?

It uses the relative altitude to the home position.

> Is it normal that the local_position.z doesn't reset to 0 when arming?

local_position is initialized when the EKF initializes but not reset on arming.

> Why those offsets? Is it just the baro drifting?

Not sure. 

> ... the copter never reaches the desired waypoint because the reference and the altitude used to check the acceptance radius differ of that offset.

The waypoints should still be reached. Are you sure about this? And can you reproduce it in SITL? Or do you have a logfile to show it?

julianoes, 2020-02-20:
I'm closing this because there was not follow up. Re-open if you have answers.

closed 2020-02-20 by julianoes