PX4/PX4-Autopilot #13518: USB issues on Durandal-v1

bkueng, 2019-11-19:
I'm experiencing issues with USB on the Durandal-v1 board with QGC.
Auto-connect is unreliable, depending on the QGC version/PC, it will not connect. This is the issue I get:
```
open failed "Device or resource busy" QSerialPort::SerialPortError(PermissionError) "Holybro on ttyACM0 (AutoConnect)" true
```
Looks like a timing issue, as adding a manual connection and connecting manually later on works.

Similar for the Firmware update: this never works (also with https://github.com/mavlink/qgroundcontrol/pull/8058), and fails with:
```
Error: Get Board Info: Get Board Info: Get Command Response: Timeout waiting for bytes to be available
Upgrade cancelled
```

@davids5

LorenzMeier, 2019-11-19:
Is that with the stock bootloader or the new bootloader? I have no issues in Mac OS

bkueng, 2019-11-19:
Our NuttX-based BL.

Looks like the auto-connect was caused by ModemManager - I upgraded my system and it was running. Disabling it resolved that issue, but Firmware updating is still not working.

davids5, 2019-11-19:
@LorenzMeier - we just had a look at it.

HB has there own usb VID now. I updated durandal Firmware and Bootloader with it prior to the merge to master.

 @bkueng  is now running the New (Nuttx) bootloader It is baked in and IIRC it SYS_BL_UPDATE=1.  

This it the output
```
[63249.720354] usb 1-2: new full-speed USB device number 111 using xhci_hcd
[63249.848741] usb 1-2: New USB device found, idVendor=3162, idProduct=008b, bcdDevice= 1.01
[63249.848747] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[63249.848750] usb 1-2: Product: PX4 BL Holybro Durandal Vx
[63249.848753] usb 1-2: Manufacturer: Holybro
[63249.848755] usb 1-2: SerialNumber: 0
[63249.850718] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[63254.500711] usb 1-2: USB disconnect, device number 111
[63254.500809] cdc_acm 1-2:1.0: failed to set dtr/rts
[63256.041410] usb 1-2: new full-speed USB device number 112 using xhci_hcd
[63256.173210] usb 1-2: New USB device found, idVendor=3162, idProduct=008b, bcdDevice= 1.01
[63256.173216] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[63256.173219] usb 1-2: Product: PX4 DurandalV1
[63256.173222] usb 1-2: Manufacturer: Holybro
[63256.173224] usb 1-2: SerialNumber: 0
[63256.175334] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
~~~

He connected by auto now but no joy on the FW update.

@LorenzMeier Would you please post your dmesg output from "plug in" + 10 Sec here?

LorenzMeier, 2021-06-02:
Solved a long time ago.

closed 2021-06-02 by LorenzMeier