PX4/PX4-Autopilot #13529: px4io issues on the test rack

dagar, 2019-11-19:
On the test rack (flashing via SWD) across all boards it appears to be quite common for the px4io to fail to respond.

[image]

Should we be force updating?

LorenzMeier, 2019-11-19:
No - because that’s a UX nightmare for users. I think we need to fix that for good.

julianoes, 2019-11-20:
This suggests a power issue: https://github.com/PX4/Firmware/issues/12308#issuecomment-503811665

And I remember talking to @antmerlino who reported that when booting hundreds of drones then often for some px4io would not come up and they needed to reboot.

antmerlino, 2019-11-20:
@julianoes I don't believe we have seen this issue since upgrading to v1.9, but I cannot say for certain. When we did see it, I don't think a PX4 reboot would work. We had to completely power cycle the drone for px4io to come up.

julianoes, 2019-11-22:
@antmerlino that's good data, thanks! And right I remember you saying a power cycle was needed.

JacobCrabill, 2019-11-26:
I've seen a similar issue that's easily reproducible - take a Cube on a standard carrier board + safety button, hold the safety button down, power the board, let go (or not) of the safety button, and the IO microcontroller will fail to boot (the px4io driver will attempt to initialize the DMA link between the two processors, and fail).  The init script (rcS) then attempts `px4io forceupdate`, which succeeds, and then everything works.

dagar, 2019-11-26:
Sounds like we should take a closer look, especially if the px4io is going to be around for a couple more flight controller generations.

julianoes, 2019-11-27:
> I've seen a similar issue that's easily reproducible - take a Cube on a standard carrier board + safety button, hold the safety button down, power the board, let go (or not) of the safety button, [...]

That's a feature! It's to boot up IO in bootloader mode to force an update.