PX4/PX4-Autopilot #13530: Negative Derivative Term in Rate PID Controller!

mzahana, 2019-11-19:
**Describe the bug**
In the PID implementation of the multicopter attitude rate controller (https://github.com/PX4/Firmware/blob/master/src/modules/mc_att_control/RateControl/RateControl.cpp#L83), the derivative term is negative ` - _gain_d.emult(rate_d)`,
```
Vector3f torque = _gain_p.emult(rate_error) + _rate_int - _gain_d.emult(rate_d) + _gain_ff.emult(rate_sp);
```
I don't understand why it's not a positive term just like the P and I terms!

Is this a bug, or there is an explanation for it?

Thanks.

mcsauder, 2019-11-20:
Hi @mzahana. Think of the proportional and integral terms like a gas pedal to speed up an error resolution and the derivative term like a brake pedal to keep the error from being resolved so fast that it results in overshoot. Much like pressing the gas pedal results in the opposite acceleration sign as the brake pedal, you need the derivative controller effort to act opposite from the P and I terms. The rate of change of the error is what the D term acts on, and this is so that when large errors are present and the P and I terms would otherwise lead to overshooting the desired value, the D term keeps the rate of change of the error value reasonable. This also allows higher P and I values than would otherwise be possible. Hopefully this makes sense!

closed 2019-11-20 by mcsauder

mzahana, 2019-11-20:
Thanks @mcsauder for the explanation. I am still thinking from a mathematical point of view. There are different implementations of discrete PID. The simplest, approximates the integral and derivative terms by first order approximations. Then, the terms are added together (after multiplying by corresponding gains). I still don't understand the negative sign in the derivative term.

mcsauder, 2019-11-20:
Hi @mzahana , Have a look at how the rates derivative is being calculated:

```
if (dt > FLT_EPSILON) {
		rate_d = (rate_filtered - _rate_prev_filtered) / dt;
	}
```

If the computation were instead the previous - current, then, as you are thinking, the sign of the derivative contribution would need be positive.  It can formulated either way, it's mostly a matter of how the rate of change is calculated.  It is also fairly straightforward to prove out in a real system or simulation that the math is correct by forcing a negative D gain to be saved and observing the resulting unstable control behavior, but following the math should also lead you to the same conclusion.  Hopefully this helps.!

RomanBapst, 2019-11-20:
@mzahana Normally you are used to seeing a positive sign as the D gain is multiplied by a pre-computed rate error. The reason why we don't use the error but just the rate is because it avoids introducing a zero into the system. I'm speaking about a zero in terms of control systems (https://en.wikibooks.org/wiki/Control_Systems/Poles_and_Zeros).
If you write the equation down as follows:
`float rate_error = rate_sp - rate`

Now zero `rate_sp` in order to avoid a zero introduced.

`float rate_error = 0 - rate = -rate`

This is where the negative sign comes from.

mzahana, 2019-11-20:
Thanks @mcsauder @RomanBapst. Now it makes sense!
I am asking this because I am doing system identification and LQR-based PID tuning. So, I needed to know what PID implementation is done in the attitude/rate loops.

So just to make sure that I am designing gains for the right controller, is there anything else that is different from a traditional PID implementation?

Cheers.