PX4/PX4-Autopilot #13541: Durandal camara trigger pin support (for lost AUX6)

bys1123, 2019-11-20:
Durandal does not have AUX6, so need to find a new way to set TRIG_PINS parameter.

bys1123, 2020-02-24:
Still need this.

bys1123, 2020-05-24:
(っ•̀ω•́)っ✎⁾⁾

hamishwillee, 2020-08-23:
@bys1123 I would have called this a feature request, not a bug.

As I [understand it](http://docs.px4.io/master/en/peripherals/camera.html) you can use any two of the AUX pins - so you can use durandal can camera triggering. The limitation is only that you can't then also use the other AUX pins. So for anything that doesn't need AUX pins - e.g. multicopters you're fine.

Obviously this does limit durandal for camera triggering and VTOLs.

I doubt this will get fixed by the core dev team, but they might take a PR :-)

bys1123, 2020-11-18:
@hamishwillee I don't know why I opened this now, closed.

closed 2020-11-18 by bys1123