PX4/PX4-Autopilot #13546: Run SIT-PX4 on WSL with Gazebo and ROS

tensarflow, 2019-11-21:
Hello there!

Unfortunately I couldn't find any resources on how to use the SITL Gazebo Simulation on WSL and get the GUI from an xserver.
I have win10 with ubuntu 18.04 WSL.
The environment toolchain is all built without problems.
Gazebo on its own runs fine. I can use the GUI through the xserver. When I try to start PX4-SITL-Gazebo with `sudo make px4_sitl gazebo` i get this output:

```
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/ensar/src/Firmware/build/px4_sitl_default/tmp && /home/ensar/src/Firmware/Tools/sitl_run.sh /home/ensa...ld/px4_sitl_default/bin/px4 none gazebo none /home/ensar/src/Firmware /home/ensar/src/Firmware/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /home/ensar/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
src_path: /home/ensar/src/Firmware
build_path: /home/ensar/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/ensar/src/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/ensar/src/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/home/ensar/src/Firmware/build/px4_sitl_default/build_gazebo
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 169.254.40.50
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
shared memfd open() failed: Function not implemented
Home directory not accessible: Permission denied
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': No such file or directory
[Err] [OpenAL.cc:84] Unable to open audio device[default]
 Audio will be disabled.
[Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
[Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[0] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[1] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[2] no joint control will be performed for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[3] no joint control will be performed for this channel.
[Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::propeller_joint] not found for channel[4] no joint control for this channel.
[Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_left_joint] not found for channel[5] no joint control for this channel.
[Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_right_joint] not found for channel[6] no joint control for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[7] no joint control will be performed for this channel.
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
SITL COMMAND: "/home/ensar/src/Firmware/build/px4_sitl_default/bin/px4" "/home/ensar/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/ensar/src/Firmware"/test_data
INFO  [px4] Creating symlink /home/ensar/src/Firmware/ROMFS/px4fmu_common -> /home/ensar/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.
[... 49 lines cut ...]
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
[Wrn] [GuiIface.cc:120] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
ERROR [px4] PX4 daemon not running yet
Error: no autostart file found (etc/init.d-posix/ERROR [px4] PX4 daemon not running yet_*)
ERROR [px4] Startup script returned with return value: 256
pxh> [Err] [REST.cc:205] Error in REST request
```

Essentially I am trying to avoid to install ubuntu on my machine, since I have not much space left. Is it worth trying? Or should I rather jump to full ubuntu?

julianoes, 2019-11-25:
You can try using Cygwin: https://dev.px4.io/master/en/setup/dev_env_windows_cygwin.html

FYI @MaEtUgR

tensarflow, 2019-11-25:
Thanks. I'll try that.

hamishwillee, 2019-11-25:
Gazebo and ROS aren't set up for Cygwin Windows build (yet), so this won't work. For Bash on Windows I only tested JMAVSim. 

When I have to use Gazebo I use the VM toolchain: https://dev.px4.io/master/en/setup/dev_env_windows_vm.html

Note though, that Gazebo is really too heavyweight for use in a VM, so I always run it headless: https://dev.px4.io/master/en/simulation/gazebo.html#headless-mode

hamishwillee, 2019-11-25:
PS This isn't really a "PX4" problem - neither Gazebo or ROS are supported on Windows. There are new versions in the pipeline that will eventually work though (according to the ROS guys and PX4 Dev Days 2019)

lovettchris, 2019-12-18:
This is now fixed by [this commit](https://github.com/PX4/Firmware/commit/01818b505f898f33176bf90f9563e84942ea56cf).

closed 2019-12-19 by bkueng

tensarflow, 2019-12-19:
Wow, huge thanks! I'll try it as soon as possible. Is this commit pushed to the master? Can I clone the standard repo to get this?

bkueng, 2019-12-19:
Yes you can use master.