PX4/PX4-Autopilot #13547: [Feature Request] Support SPL06-001 High Precision (5cm) Barometer

bys1123, 2019-11-21:
There is a Digital pressure sensor become more and more popular in China, model number SPL06-001, manufactured by GoerTek.
It have 5cm relative accuracy, can make autopilot have better altitude performance.

Module Purchase: https://www.banggood.com/CJMCU-06-SPL06-001-Drone-Pressure-Height-Sensor-Module-Position-5cm-p-1295458.html
Datasheet: https://1drv.ms/b/s!AmscxcpJYtuYlle8d1lGb1rDLp7h

Kirishikesan, 2020-03-09:
@bys1123 , I would like to take up this issue and write a driver for the SPL06-007 High precision low cost barometer.

bys1123, 2020-03-09:
@Kirishikesan That'll be awesome, do you already have this baro component?

Kirishikesan, 2020-03-10:
No, I do not have it with me yet. I tried searching for those sensors locally but I could not find them. I'll get that barometer from online asap. Thank you @bys1123.

bys1123, 2020-04-28:
We created this general usage driver in C first: https://github.com/DimianZhan/spl06
@Kirishikesan

Ncerzzk, 2022-02-19:
hey, I have implement the driver and will commit it recently. While I only tested on linux, there may be some problems(or no) on nuttx. If someone could test it on nuttx, it would be very nice.