PX4/PX4-Autopilot #13550: How to get pose groundtruth from gazebo sitl?

wuxianshen, 2019-11-21:
I want to get pose groundtruth from gazebo sitl, like the local NED groundtruth and so on.
I follow the instructions of gazebo topics, use "gz topic -l" to show the msgs available in gazebo when running px4_sitl, which shows some iris (the sitl model) related topics:
[image]
And then I try to read these topics through libgazeboX-dev (instructions from http://gazebosim.org/tutorials?tut=topics_subscribed&cat=transport), but the content cannot be decoded with protobuf, like:
[image]
So how can I get these information from gazebo? Where to find these protobuf headers to decode
the information?

Thanks!

julianoes, 2019-11-25:
@mhkabir could you help here?

TSC21, 2019-11-25:
@wuxianshen do you want to get the groundtruth data in ROS? If yes, there's a topic you can subscribe to in the case of the Iris model - `/ground_truth` - thanks to the `odometry_plugin_macro` on the `rotors_description`:https://github.com/PX4/sitl_gazebo/blob/master/models/rotors_description/urdf/component_snippets.xacro#L555-L585

TSC21, 2019-11-25:
Also, I am not sure you are aware, but in SITL, the groundtruth is actually logged on the PX4 side as well through several `vehicle_*_groundtruth` uORB topics: https://github.com/PX4/Firmware/blob/master/src/modules/simulator/simulator_mavlink.cpp#L417-L510

closed 2019-12-05 by mhkabir