PX4/PX4-Autopilot #13551: Weird Gyro bias values on thermal compensation on graph

feemi, 2019-11-21:
**Describe the bug**
Gyro 1 has weird values when doing offboard calibration. We tried couple of different cube autopilots.

**To Reproduce**
Steps to reproduce the behavior:
1.  Do offboard calibration
2. Run process_sensor_caldata.py program

**Expected behavior**
Gyro 1 bias values should be similar to other sensors

**Log Files and Screenshots**
[image]
[log002.ulg.pdf](https://github.com/PX4/Firmware/files/3875585/log002.ulg.pdf)

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Pixhawk Cube 2.1 fmu-v3
- px4 v1.9.2

mhkabir, 2019-11-21:
This is because the Bosch sensors only report temperatures to the nearest degree. Performance should be unaffected.

feemi, 2019-11-21:
Thanks @mhkabir I am closing the issue

closed 2019-11-21 by feemi