PX4/PX4-Autopilot #13552: Wired 360 flip

mathias-h, 2019-11-21:
Could someone please take a look at this log, and maybe explain why the drone flips 360.

https://logs.px4.io/plot_app?log=46ebc350-9cb9-4eee-9892-a2f54c2aa4ed

Thank you.

mathias-h, 2019-11-22:
i guess this is not really the right place to ask this question. i will try https://discuss.px4.io/ instead

closed 2019-11-22 by mathias-h

mcsauder, 2019-11-22:
Hi @mathias-h , your roll rate estimate and set point diverge, (the autopilot wants to roll one direction and the drone rolls the other way).  It looks like the opposite control output from what is required is being applied. Verify your motor rotation directions as a first start and, (as you are doing), continue the discussion on px4 discuss.

mathias-h, 2019-11-23:
I’m pretty sure the motor directions are ok.
The drone flies fine in manual mode. 
And just before the flip it flew for about a minute just fine.
But where is the link to the discuss topic:
https://discuss.px4.io/t/quad-destroyed-360-flip-in-altitude-mode/13975