PX4/PX4-Autopilot #13558: Proposal – generate mavlink/urb commands from one xml

BazookaJoe1900, 2019-11-22:
Hi,
For long time I see a bad coding standard related to the uorb's vehicle_command and mavlink's VEHICLE_CMD_ that I think should be solved. I wanted your opinion and notes about that and if we will found it to be solved I can help doing that.

The problem reading vehicle_command parameters is not by structs, and its by reading each parameter by its 'known' position. 
For example:

The metadata is there, on the mavlink xml. But it doesn't generated or used, not in the C generators of mavlink.
I thought of generating structs from the xml metadata, using those structs when need to read/write commands, remove the double definitions of the uorb VEHICLE_CMD_. Maybe later to continue and make the parameters type protected and using enums.

what I found tricky is:

- mavlink and urb are has different structs 

- generating uorb related struct from mavlink xml might seems odd

any thoughts?

LorenzMeier, 2019-11-22:
Thanks for sharing your thoughts! Overall I like more automation, however, we made a conscious decision to implement a correct API between uORB and MAVLink in order to have a point of isolation from a cybersecurity and deployment / version control perspective. So what is not going to be acceptable are any automatic translations. What is however desirable is to generate the conversions we do and then "enable" those manually. This would be better with respect to what we are doing today as it would ensure consistency.

BazookaJoe1900, 2019-11-22:
@LorenzMeier thanks for the response
In the matter of translation, I thought of just creating structs that on one side reflect the commands structures, as in in XML. On the other 'side' its on the uorb c language structural.
When you 'know' which command you want to decode you just need to use the correct structural command. (using pointer, or something like that)

```
union mav_commands
{
    struct mav_cmd_MAV_CMD_NAV_WAYPOINT; // cmd value=16 (and other metadata...)
    struct mav_cmd_MAV_CMD_NAV_LOITER_UNLIM; // cmd value=17 (and other metadata...) 
…
}

struct mav_cmd_MAV_CMD_NAV_WAYPOINT
{
    double reserved5; 
    double reserved6; 
    float Hold; 		//metadata (description, values, enum, and so on...._) 
    float Accept_Radius; //metadata (description, values, enum, and so on...._) 
    float Pass_Radius;
    float Yaw;
    float reserved7;
};

struct mav_cmd_MAV_CMD_NAV_LAND_LOCAL
{
	double Y_Position;
	double X_Position;
	float Target;
	float Offset;
	float Descend_Rate;
	float Yaw;
	float Z_Position;
};
```

I am aware that in many ways its not 'elegant', but I think its better than the current 'hard codded' way.

In the matter of the duplication of same data in MAV_CMD and VEHICLE_CMD_:
Comparing the lists, shows that VEHICLE_CMD_ is subset of MAV_CMD. Except for some diffs, that I think should be checked (NAV_GUIDED_LIMITS , NAV_GUIDED_MASTER, NAV_PRECLAND).
Any way, making VEHICLE_CMD enum as MAV_CMD + other 'internal' command should not be problematic and will protect from duplication and conflicts.
But, as you said that has been consciously done, and I know that I don't know all the reasons that led to that.

BazookaJoe1900, 2020-01-24:
seems to be irrelevant, closing

closed 2020-01-24 by BazookaJoe1900