PX4/PX4-Autopilot #13567: Nuttx cmath: compilation error

kamilritz, 2019-11-22:
**Describe the bug**
[This commit](https://github.com/PX4/Firmware/commit/99aae8b8915078eef6d4e83d2823894339027aab) is stopping me from building the Firmware on my computer.

I get compilation errors of the following form:

```
In file included from NuttX/nuttx/include/cxx/cmath:46:0,
                 from ../../src/lib/ecl/ecl.h:99,
                 from ../../src/lib/ecl/geo/geo.cpp:46:
/usr/include/newlib/math.h:18:3: error: '__ULong' does not name a type
   __ULong i[2];
   ^~~~~~~
compilation terminated due to -Wfatal-errors.
```
Do I have to upgrade anything? Does anybody else have the same issue?
@dagar FYI

dagar, 2019-11-22:
What's your development environment and which version of the toolchain (arm-none-eabi-gcc) are you using? 

Additionally, <cmath> from NuttX is pretty incomplete. We're probably better not using it at all until we have something better in place.

kamilritz, 2019-11-22:
Using Ubuntu 18.04 and arm-none-eabi-gcc version  6.3.1

dagar, 2019-11-22:
You should update to GCC 7 from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads or run this script in tree (https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh).

You can often get away with running different versions of the toolchain, but we're not in a position to support that. Separately I think we should also remove most of the <cmath> inclusions.

We'll hopefully be jumping to GCC 9 soon. https://github.com/PX4/Firmware/pull/13448

kamilritz, 2019-11-22:
Thank you very much. I had an old arm-none-eabi-gcc version installed that I had to remove first. After that the toolchain installation did the rest. Thanks I can build current master again

closed 2019-11-22 by kamilritz

JeffC0227, 2020-01-04:
Using Ubuntu 16.04LTS and gcc-arm-none-eabi-4_9-2015q3-20150921-linux, i meet same problem

julianoes, 2020-02-14:
We only support this version:
https://github.com/PX4/Firmware/blob/342e0da7961f9a3301706ed3835cfc163b14b2ed/Tools/setup/ubuntu.sh#L141
Please use this script to update your system.

kashishdhal, 2020-03-04:
Hello I am having the same issue:

/usr/include/newlib/math.h:18:3: error: '__ULong' does not name a type
   __ULong i[2];
   ^~~~~~~

Upon checking the versin of gcc by the following command I get:
$ arm-none-eabi-gcc --version

I get:

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

which means the version of the gcc is as desired.

I am using Ubuntu 18.04 operating system. Any help will be appreciated.

Regards,
Kashish Dhal

julianoes, 2020-03-04:
@kashishdhal 

> You should update to GCC 7 from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads or run this script in tree (https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh).

kashishdhal, 2020-03-04:
@julianoes 
I already checked the gcc version using $ arm-none-eabi-gcc --version.

It is GCC 7-2017-q4-major. Is that not correct?

DO I still need to do a reinstall?

kashishdhal, 2020-03-04:
@julianoes 

As per your suggestion, upon running the script in tree (which I did before as well) I get following:

E: Unable to locate package gdb-arm-none-eabi
Installing GCC to: /home/gnclab/gcc-arm-none-eabi-7-2017-q4-major
 GCC already installed.
RESTART YOUR COMPUTER to complete installation of PX4 development toolchain

Since **gdb-arm-none-eabi** seems to be missing, I installed it manually from arm website (https://packages.ubuntu.com/xenial/gdb-arm-none-eabi) but it needed a lot of dependencies so I installed them one by one manually. However, while installing **gdb-arm-none-eabi** I get the error which is:

gpg: skipped "tin@debian.org": No secret key
gpg: dpkg-sign.5s8QG8jk/gdb-arm-none-eabi_9_amd64.buildinfo: clear-sign failed: No secret key
dpkg-buildpackage: error: failed to sign .buildinfo file

I thought this is just a signature error and should be fine.

Please let me know what am I doing wrong here?

Regards,
Kashish Dhal

julianoes, 2020-03-05:
> Since gdb-arm-none-eabi seems to be missing, I installed it manually from arm website 

It's not missing, check `which arm-none-eabi-gdb`.

kashishdhal, 2020-03-05:
It returns:

/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gdb

kashishdhal, 2020-03-06:
@julianoes 

Hello, please let me know what should I do next to correct it? I am not able to run make command. I already showed you the error I am getting in the above message. Please help. I will appreciate your help.

@kamilritz 

Since you were able to fix it, do you have any suggestions? Please help.

hamishwillee, 2020-03-06:
@kamilritz Now that you are using the correct gcc, what build command are you using? 

Do a `make distclean` and then try to rebuild. What error, if any are you seeing?

kashishdhal, 2020-03-06:
Hello @hamishwillee 

I am still getting the same error:

/usr/include/newlib/math.h:18:3: error: '__ULong' does not name a type
   __ULong i[2];

hamishwillee, 2020-03-06:
@kashishdhal I assume you are building the master codeline?

I just tried this on a vanilla Ubuntu 18.04.4 using the setup script ubuntu.sh. All worked rather nicely.
[image]

Can you put your full build output into a gist and link here. That might help @julianoes advise - I'm a bit out of my depth.

kashishdhal, 2020-03-06:
I am using the same Firware directory which was used for SITL so maybe I will create another Firmware directory and try to build again from there. If still unsuccessful, will post the code here as you asked.

julianoes, 2020-03-06:
You need to remove the toolchain that you have installed again, probably:
```
sudo apt-get remove gdb-arm-none-eabi gcc-arm-none-eabi
```

kashishdhal, 2020-03-06:
So when I **bash ubuntu_sim_nuttx.sh** , I get the following:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/Firmware2/Firmware</b></font>$ sudo bash ubuntu_sim_nuttx.sh 
[sudo] password for gnclab: 
Downloading dependent script &apos;ubuntu_sim.sh&apos;
--2020-03-06 14:47:16--  https://raw.githubusercontent.com/PX4/Devguide/v1.9.0/build_scripts/ubuntu_sim.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.48.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.48.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1576 (1.5K) [text/plain]
Saving to: ‘STDOUT’

-                   100%[===================&gt;]   1.54K  --.-KB/s    in 0s      

2020-03-06 14:47:16 (252 MB/s) - written to stdout [1576/1576]

Downloading dependent script &apos;ubuntu_sim_common_deps.sh&apos;
--2020-03-06 14:47:16--  https://raw.githubusercontent.com/PX4/Devguide/v1.9.0/build_scripts/ubuntu_sim_common_deps.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.48.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.48.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2839 (2.8K) [text/plain]
Saving to: ‘STDOUT’

-                   100%[===================&gt;]   2.77K  --.-KB/s    in 0s      

2020-03-06 14:47:16 (10.0 MB/s) - written to stdout [2839/2839]

We must first remove modemmanager
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package &apos;modemmanager&apos; is not installed, so not removed
The following packages were automatically installed and are no longer required:
  binutils-arm-none-eabi libgraphviz-dev libgvc6-plugins-gtk
  libnewlib-arm-none-eabi libnewlib-dev libopenal-dev libqtpropertybrowser4
  libstdc++-arm-none-eabi-newlib libxdot4
Use &apos;sudo apt autoremove&apos; to remove them.
0 upgraded, 0 newly installed, 0 to remove and 126 not upgraded.
Installing common dependencies
Hit:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic InRelease
Hit:2 http://linux.teamviewer.com/deb stable InRelease                         
Hit:3 http://ppa.launchpad.net/system76-dev/stable/ubuntu bionic InRelease     
Hit:4 http://archive.ubuntu.com/ubuntu bionic InRelease                        
Hit:5 http://security.ubuntu.com/ubuntu bionic-security InRelease              
Hit:6 http://packages.ros.org/ros/ubuntu bionic InRelease                
Hit:7 http://archive.ubuntu.com/ubuntu bionic-updates InRelease          
Reading package lists... Done                      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting &apos;libimage-exiftool-perl&apos; instead of &apos;exiftool&apos;
build-essential is already the newest version (12.4ubuntu1).
zip is already the newest version (3.0-11build1).
astyle is already the newest version (3.1-1ubuntu2).
genromfs is already the newest version (0.5.2-2build3).
libimage-exiftool-perl is already the newest version (10.80-1).
ninja-build is already the newest version (1.8.2-1).
qtcreator is already the newest version (4.5.2-3ubuntu2).
git is already the newest version (1:2.17.1-1ubuntu0.5).
cmake is already the newest version (3.10.2-1ubuntu2.18.04.1).
The following packages were automatically installed and are no longer required:
  binutils-arm-none-eabi libgraphviz-dev libgvc6-plugins-gtk
  libnewlib-arm-none-eabi libnewlib-dev libopenal-dev libqtpropertybrowser4
  libstdc++-arm-none-eabi-newlib libxdot4
Use &apos;sudo apt autoremove&apos; to remove them.
0 upgraded, 0 newly installed, 0 to remove and 126 not upgraded.
/usr/bin/xxd
Reading package lists... Done
Building dependency tree       
[... 155 lines cut ...]
  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable &quot;CC&quot; or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:190 (project):
  The CMAKE_ASM_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable &quot;ASM&quot; or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also &quot;/home/gnclab/Firmware2/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeOutput.log&quot;.
See also &quot;/home/gnclab/Firmware2/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeError.log&quot;.
Error: /home/gnclab/Firmware2/Firmware/build/px4_fmu-v4_default is not a directory
Makefile:198: recipe for target &apos;px4_fmu-v4_default&apos; failed
make: *** [px4_fmu-v4_default] Error 1
</pre>

kashishdhal, 2020-03-07:
Hello @julianoes  and @hamishwillee 

This time, I removed  GCC compiler and also the ~/Firmware and ~/src/Firmware directory. After that I removed the path of GCC from ~/.profile. Then restarted the computer the computer. Upon cloning the Firmware directory again and running ubuntu_nuttx.sh, I got no error.

However, when I build this time, I get the following:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/Firmware</b></font>$ sudo make px4_fmu-v4_default
[sudo] password for gnclab: 
-- PX4 version: v1.11.0-beta1-572-gcf47c2155f
-- PX4 config file: /home/gnclab/Firmware/boards/px4/fmu-v4/default.cmake
-- PX4 config: px4_fmu-v4_default
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is unknown
-- The C compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: arm-none-eabi-gcc
CMake Error at CMakeLists.txt:195 (project):
  The CMAKE_CXX_COMPILER:

    arm-none-eabi-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable &quot;CXX&quot; or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:195 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable &quot;CC&quot; or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:195 (project):
  The CMAKE_ASM_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable &quot;ASM&quot; or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also &quot;/home/gnclab/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeOutput.log&quot;.
See also &quot;/home/gnclab/Firmware/build/px4_fmu-v4_default/CMakeFiles/CMakeError.log&quot;.
Error: /home/gnclab/Firmware/build/px4_fmu-v4_default is not a directory
Makefile:198: recipe for target &apos;px4_fmu-v4_default&apos; failed
make: *** [px4_fmu-v4_default] Error 1
</pre>

Now this is something new but to solve this I set the path of gcc in ~/.profile, by adding the following commands:
export PATH=/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin:$PATH
export PATH=/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc:$PATH
export PATH=/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++:$PATH

Followed by restarting the computer.

I also added the following lines in the CMakelist.txt:
set(ASM "/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc")
set(CXX "/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++")
set(C "/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc")

But I am not able to resolve the error. Please help.

Regards,
Kashish Dhal

kashishdhal, 2020-03-07:
Hello @julianoes and @hamishwillee 

I was successfully able to build it this time.

Here is what I was doing wrong:

1. Your bash script was downloading Firmware again into ~/src/Firmware However, I already cloned that and was working on ~/Firmware. 

2. environment path for gcc is set but when you type sudo, environment variables changes and the path is not there anymore. Solution is to use sudo in the whole terminal using **sudo -s**. Then afterwards run this command **export PATH=/home/gnclab/gcc-arm-none-eabi-7-2017-q4-major/bin:$PATH** which will again set the path of gcc we want to use.

After fixing these two errors, it was build successfully.

Thanks to the community for their help. A suggestion for them,  since PX4 is used quite a bit among aerospace community and they do not have enough understanding of compiling the packages. These things should be simplified.

Regards,
Kashish Dhal

julianoes, 2020-03-09:
Did you do as it tells you at the end of the script?
```
RESTART YOUR COMPUTER to complete installation of PX4 development toolchain
```

kashishdhal, 2020-03-09:
Yes I did. That did not help. 

After you restart your computer you get path of GCC set up in environment. But if you execute make command with sudo. You won't see that path and it will complain gcc not found.

julianoes, 2020-03-10:
> But if you execute make command with sudo. You won't see that path and it will complain gcc not found.

Why do you ever use sudo to build? That should not be required. And if you used it once then you will have build files owned by root, so you need to clean up with sudo before the permission errors go away.

Try:
```
sudo make clean
```
or:
```
sudo rm -rf build
```
And then build without sudo again.

kashishdhal, 2020-03-10:
I guess you are right, every time I have to access the firmware directory, I have to use sudo now. I installed Qt creater and it can only access the files when launched from terminal with sudo command.

I did that because it was not letting me make saying permission denied. So you are saying that I should clean the firmware directory using the sudo make clean and then rebuild by make px4...... and then it won't complain about permissions?

Also, another question I like to modify the firmware to accept the individual motor commands for quad, I saw on PX4 website that I need to modify mixer file but it was not much clear to me.
Do you know how to do that? If yes, would you be able to help a bit.

Regards,
Kashish Dhal

julianoes, 2020-03-11:
> So you are saying that I should clean the firmware directory using the sudo make clean and then rebuild by make px4

That is what I wrote above.

For other questions, please use https://discuss.px4.io or Slack, and check the docs:
https://dev.px4.io/master/en/concept/mixing.html

kashishdhal, 2020-03-11:
Hello @julianoes ,

I did what you said and now I am getting the following:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/src/Firmware</b></font>$ make px4_fmu-v4
[0/1284] git submodule src/drivers/uavcan/libuavcan
[1/1284] Generating mixer_multirotor_6dof.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_6dof.generated.h 
cd /home/gnclab/src/Firmware/build/px4_fmu-v4_default/src/lib/mixer/MultirotorMixer &amp;&amp; /usr/bin/python3 /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_6dof.generated.h
Failed to import toml: No module named &apos;toml&apos;

You may need to install it using:
    pip3 install --user toml

[3/1284] Generating mixer_multirotor_normalized.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_normalized.generated.h 
cd /home/gnclab/src/Firmware/build/px4_fmu-v4_default/src/lib/mixer/MultirotorMixer &amp;&amp; /usr/bin/python3 /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --normalize -f /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
Failed to import toml: No module named &apos;toml&apos;

You may need to install it using:
    pip3 install --user toml

[4/1284] Generating mixer_multirotor.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor.generated.h 
cd /home/gnclab/src/Firmware/build/px4_fmu-v4_default/src/lib/mixer/MultirotorMixer &amp;&amp; /usr/bin/python3 /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py -f /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/gnclab/src/Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/gnclab/src/Fir [... cut ...] xecuting pip with sudo, you may want sudo&apos;s -H flag.</font>
<font color="#C4A000">The directory &apos;/home/gnclab/.cache/pip&apos; or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo&apos;s -H flag.</font>
Requirement already satisfied: toml in /home/gnclab/.local/lib/python3.6/site-packages</pre>

**Is there anything else I would have to remove now?**

julianoes, 2020-03-20:
It looks like your Python or Pip installation is screwed up.

Please paste the output of these commands:
```
which python
which python3
which pip
which python3
python --version
python3 --version
pip --version
pip3 --version
```

kashishdhal, 2020-03-24:
<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ which python
/usr/bin/python
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ which python3
/usr/bin/python3
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ which pip
/usr/local/bin/pip
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ which python3
/usr/bin/python3
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ python --version
Python 2.7.17
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ python3 --version
Python 3.6.9
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ pip --version
pip 20.0.2 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
<font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)
</pre>

julianoes, 2020-03-25:
Sorry, I forgot `which pip3`.

But I would clean out all pip and python things from `/usr/local` and then do:
```
sudo apt-get reinstall python python3 python-pip python3-pip
```

kashishdhal, 2020-03-25:
Hello @julianoes 

Here is the output:
<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~</b></font>$ which pip3
/usr/bin/pip3
</pre>

Now should I manually delete all these files?

Regards,
Kashish Dhal

julianoes, 2020-03-26:
I would suggest to remove everything related to Python in `/usr/local` unless you have a good reason for it. Then I would run the `ubuntu.sh` script again which now should work fine.

kashishdhal, 2020-03-30:
Hello @julianoes ,

I did what was suggested and additionally removed python stuff from /usr/.local. However, make still fails with following error:

<pre>ccache: error: Failed to create temporary file for /home/gnclab/.ccache/tmp/tmp.cpp_stderr: Permission denied
</pre>

julianoes, 2020-03-30:
Did you do a `make clean` And can you paste the whole build output?

kashishdhal, 2020-03-30:
Here is the whole build output:

<pre><font color="#8AE234"><b>gnclab@system76-pc</b></font>:<font color="#729FCF"><b>~/src/Firmware</b></font>$ make px4_fmu-v4_default
[15/1152] Generating nuttx/fs/libfs.a
FAILED: NuttX/nuttx/fs/libfs.a 
cd /home/gnclab/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx &amp;&amp; find fs -type f -name *.o -delete &amp;&amp; make -C fs --quiet --no-print-directory all TOPDIR=/home/gnclab/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx KERNEL=y EXTRADEFINES=-D__KERNEL__
ccache: error: Failed to create temporary file for /home/gnclab/.ccache/tmp/tmp.cpp_stderr: Permission denied
Makefile:97: recipe for target &apos;fs_initialize.o&apos; failed
make[1]: *** [fs_initialize.o] Error 1
[16/1152] Generating nuttx/libs/libc/libc.a
FAILED: NuttX/nuttx/libs/libc/libc.a 
cd /home/gnclab/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx &amp;&amp; find libs/libc -type f -name *.o -delete &amp;&amp; make -C libs/libc --quiet --no-print-directory all TOPDIR=/home/gnclab/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx KERNEL=n EXTRADEFINES=
ccache: error: Failed to create temporary file for /home/gnclab/.ccache/tmp/tmp.cpp_stderr: Permission denied
Makefile:158: recipe for target &apos;bin/arch_memcpy.o&apos; failed
make[1]: *** [bin/arch_memcpy.o] Error 1
[24/1152] Generating apps/builtin/registry/px4_sih_main.bdat
ninja: build stopped: subcommand failed.
Makefile:198: recipe for target &apos;px4_fmu-v4_default&apos; failed
make: *** [px4_fmu-v4_default] Error 1
</pre>

It looks to me that my permissions are messed up. Let me do a make clean

kashishdhal, 2020-03-31:
Hello @julianoes ,

It worked this time! Thanks for the help!

kashishdhal, 2020-03-31:
Hello @julianoes,

Since I am able to build the firmware now, I would like to know that what is the easiest way to get started to play with firmware files to edit them for my desired missions. I would like to learn the whole architecture so that I can edit the firmware when needed.

Also, I am running into another issue since last one month. I am trying to push the VICON data through /mavros/vision_pose/pose as specified in the https://dev.px4.io/v1.9.0/en/ros/external_position_estimation.html and when I echo /mavros/local_position/pose I see the data given by vicon for atleat 1-2 min after every rebbot.

However, after 1 min of reboot local pose shows x = 0; y =0 and z  some random value. Also, when I am pushing the vicon data, the orientation is never getting updated. Please let me know if you would be able to help.
Should I just edit EKF2 file to read vicon data for now? 

Regards,
Kashish Dhal

julianoes, 2020-03-31:
Please don't add to an issue.
And for questions ask in discuss.px4.io.

kashishdhal, 2020-03-31:
Okay thanks @julianoes

bigbellmercy, 2020-05-06:
From this issue, I got to know that 'Firmware/Tools/setup/ubuntu.sh' should be executed before to build PX4 v1.11.0 and above.